filipsPL/autowx2 #15: KeyError: 'fixedTime'

SA7BNT, 2018-10-02:
Hallo, after 2 weeks nonstop running im getting this error in the morning. I have updated the keppler but still having this issue. 

sa7bnt@owncloud:~/autowx2$ python autowx2.py
2018-10-02 08:03   Reading the default dongle shift 
2018-10-02 08:03   Recently used dongle shift is: 11.109 ppm 
Traceback (most recent call last):
  File "autowx2.py", line 283, in <module>
    passTable = genPassTable()
  File "autowx2.py", line 125, in genPassTable
    cron = satellitesData[satellite]['fixedTime']
KeyError: 'fixedTime'
sa7bnt@owncloud:~/autowx2$

Whats the problem ??

filipsPL, 2018-10-02:
Hi,
thanks for the report. Was anything changed in the config file (`autowx2_conf.py`)?
Is the tle file valid (check the size with `ls -la var/tle/all.txt`)?
Did you update the system recently?

F.

SA7BNT, 2018-10-02:
Hi,
Nothing was changed in the (autowx2_conf.py)
No system update at the last week. 
And thats the output from (ls -la var/tle/all.txt)

sa7bnt@owncloud:~/autowx2$ ls -la var/tle/all.txt
-rw-r--r-- 1 sa7bnt sa7bnt 80476 Okt  2 08:02 var/tle/all.txt

sa7bnt@owncloud:~/autowx2$ python autowx2.py
2018-10-02 13:46   Reading the default dongle shift 
2018-10-02 13:46   Recently used dongle shift is: 11.109 ppm 
Traceback (most recent call last):
  File "autowx2.py", line 283, in <module>
    passTable = genPassTable()
  File "autowx2.py", line 125, in genPassTable
    cron = satellitesData[satellite]['fixedTime']
KeyError: 'fixedTime'
sa7bnt@owncloud:~/autowx2$ 

How can i get it at work again??

SA7BNT, 2018-10-02:
[autowx2_conf.py.zip](https://github.com/filipsPL/autowx2/files/2438283/autowx2_conf.py.zip)

Hope that will help

filipsPL, 2018-10-02:
Thanks! However I can't reproduce the problem... Could you please:

- try running `genpasstable.py` - it should generate the same error
- try fetch TLE data once more. Does the problem remain?
- try to use the attached tle data [all.txt](https://github.com/filipsPL/autowx2/files/2438340/all.txt) - put in into `var/tle/`. Does the problem remain?

SA7BNT, 2018-10-02:
genpasstable.py getting the same issue
> sa7bnt@owncloud:~/autowx2$
sudo python genpasstable.py 
Traceback (most recent call last):
  File "genpasstable.py", line 194, in <module>
    passTable = genPassTable(howmany=50)
  File "/home/sa7bnt/autowx2/autowx2.py", line 125, in genPassTable
    cron = satellitesData[satellite]['fixedTime']
KeyError: 'fixedTime'
sa7bnt@owncloud:~/autowx2$ python genpasstable.py 
Traceback (most recent call last):
  File "genpasstable.py", line 194, in <module>
    passTable = genPassTable(howmany=50)
  File "/home/sa7bnt/autowx2/autowx2.py", line 125, in genPassTable
    cron = satellitesData[satellite]['fixedTime']
KeyError: 'fixedTime'
sa7bnt@owncloud:~/autowx2$

After running update-keps.sh still the same issue

And after replace the all.txt from git and chmod it to 755 and run python autowx2.py still same issue.

So if i get it right the problem is that he can´t read the all.txt (TLE) 

sa7bnt@owncloud:~/autowx2$ ls -la var/tle/all.txt
-rwxr-xr-x 1 sa7bnt sa7bnt 72580 Okt  2 15:24 var/tle/all.txt
sa7bnt@owncloud:~/autowx2$

SA7BNT, 2018-10-02:
UPDATE: 
rm all autowx2 
git clone new autowx2 
config autowx2_conf.py (by hand NO backup copy)
./configure.sh 
Test python autowx2.py (is´s running without issue)
running /bin/update-keps.sh (as user)
Getting issue again
sa7bnt@owncloud:~/autowx2$ python autowx2.py
2018-10-02 17:47   Reading the default dongle shift 
2018-10-02 17:47   Recently used dongle shift is: 11.248 ppm 
Traceback (most recent call last):
  File "autowx2.py", line 283, in <module>
    passTable = genPassTable()
  File "autowx2.py", line 125, in genPassTable
    cron = satellitesData[satellite]['fixedTime']
KeyError: 'fixedTime'
sa7bnt@owncloud:~/autowx2$
sa7bnt@owncloud:~/autowx2/var/tle$ ls -lisa
insgesamt 172
1576481  4 drwxr-xr-x 2 sa7bnt sa7bnt  4096 Okt  2 17:45 .
1576474  4 drwxr-xr-x 3 sa7bnt sa7bnt  4096 Okt  2 17:36 ..
1576482 80 -rw-r--r-- 1 sa7bnt sa7bnt 80476 Okt  2 17:45 all.txt
1576483 12 -rw-r--r-- 1 sa7bnt sa7bnt 10920 Okt  2 16:05 amateur.txt
1576484 28 -rw-r--r-- 1 sa7bnt sa7bnt 25872 Okt  2 16:05 cubesat.txt
1576485 36 -rw-r--r-- 1 sa7bnt sa7bnt 35788 Okt  1 15:46 multi.txt
1576486  8 -rw-r--r-- 1 sa7bnt sa7bnt  7896 Okt  2 16:05 weather.txt
sa7bnt@owncloud:~/autowx2/var/tle$
[all.txt.zip](https://github.com/filipsPL/autowx2/files/2439062/all.txt.zip)

SA7BNT, 2018-10-02:
I got the issue. The NOAA tle data has changed. 

You have 
NOAA-19
and in the download tle the NOAA tle is like 
NOAA 19 [+]
http://www.celestrak.com/NORAD/elements/noaa.txt

filipsPL, 2018-10-02:
Indeed! :+1: They've changed the NOAA notation in the weather.txt:
```bash
$ cat all.txt | grep NOAA
NOAA 15                 
NOAA 18                 
NOAA 19                 
NOAA 20
```

now one has to adjust the `autowx2_conf.py` to the new nomenclature, i.e.:

```python
    'NOAA 18': {
        'freq': '137912500',
        'processWith': 'modules/noaa/noaa.sh',
        'priority': 2},
    'NOAA 15': {
        'freq': '137620000',
        'processWith': 'modules/noaa/noaa.sh',
        'priority': 2},
    'NOAA 19': {
        'freq': '137100000',
        'processWith': 'modules/noaa/noaa.sh',
        'priority': 2},
```

I will update the code soon.

closed 2018-10-03 by filipsPL