filipsPL/autowx2 #19: Still NOAA prblem

SA7BNT, 2018-10-03:
Right now, all NOAA records will be saved in autowx2. Don't know why. All other records seems to be fine.
Something you can confirm?

filipsPL, 2018-10-03:
It was because  the satellite name now contains spaces. It should be fixed now f7ab129. Please `git pull` the new code and let me know if this works for you.

SA7BNT, 2018-10-03:
Seems still to be something strange in the output. 

> 
2018-10-03 21:26   Reading the default dongle shift 
2018-10-03 21:26   Recently used dongle shift is: 11.269 ppm 
2018-10-03 21:26   Next five passes: 
2018-10-03 21:26   ?    NOAA 19 :: 2018-10-04 04:33 to 2018-10-04 04:48, dur: 15:00, max el. 64°; azimuth: 294° (WNW?7m2018-10-04 14:28 to 2018-10-04 14:43, dur: 15:00, max el. 72°; azimuth: 254° (WSW?guration file...
2018-10-03 21:29:01,433 INFO:Config - Tested SDR #00000002 OK
 * Serving Flask app "autorx.web" (lazy loading)
2018-10-03 21:29:01,436 INFO:Started Flask server on http://0.0.0.0:5000
 * Environment: production
   WARNING: Do not use the development server in a production environment.
2018-10-03 21:29:01,437 INFO:Telemetry Logger - Started Telemetry Logger Thread.
   Use a production WSGI server instead.
2018-10-03 21:29:01,437 INFO:E-Mail - Started E-Mail Notifier Thread
 * Debug mode: off
2018-10-03 21:29:01,510 INFO:APRS-IS - APRS Uploader Started.
2018-10-03 21:29:01,513 INFO:SDR #00000002 has been allocated to Scanner.

First NOAA pass will be 04:33 UTC

filipsPL, 2018-10-04:
Do you mean:
> First **NOAA** pass will be 04:33 UTC
?

SA7BNT, 2018-10-04:
No no.
When you start the autowx2.py in terminal, at the start it always show the next 5 passes and the dongle calibrate and when the next pass will be (in min). 
Now i just get one pass in the output and nothing about the dongle calibrate or when the next track will be. 
[image]

BTW: I´m getting img. from NOAA again. All rec. are there where they should be.

Thanks for help.

filipsPL, 2018-10-08:
That is strange. I have to investigate it further...

filipsPL, 2018-10-08:
The NOAA satellite name has changed again (to NOAA-xx version again). I've added the check for the name of the satellite in the code. If the satellite name is not found neither in the TLE data nor has the `fixedTime` set, the info is printed. This is available in the newest version  2ac40c7.

Please let me know if now it works for you.

filipsPL, 2018-10-09:
>  Is it possible to get some logfile?? Have not seen it in the config. Would be nice. Or something like e-mail notify if something goes wrong

Please create a new issue,  I will handle this :)
-- Wysłane z mojego ZX Spektrum.