filipsPL/autowx2 #22: ISS recording problem

SA7BNT, 2018-10-16:
Hey,
today i record the ISS and i observe that the record audio is to fast in its audio.
Here do you hear the Original voice of Alexander Gerst.
https://soundcloud.com/florian-df2et/ariss-school-conact-with-astro_alex-16102018-1028-utc
And here can you hear what autowx2 was recording.
[20181016-1207_ISS.mp3.zip](https://github.com/filipsPL/autowx2/files/2483352/20181016-1207_ISS.mp3.zip)
Something you have observed earlier??

Greeting 
Tim

filipsPL, 2018-10-16:
I think this could be a matter of setting for mp3 encoding. I had a similar issues before, but I don't know how to fix it. I guess one need to play a little bit with mp3 encoding settings to get the proper speed/quality of the recording...

SA7BNT, 2018-10-17:
I tried that and it sounds a lot better.
I add a k after -s 32 and add -l 0 in the front

timeout $duration rtl_fm -f $freq -M wbfm -g 49.6 -l 0 | lame -r -s 32k -m m - $recdir/$fileNameCore.mp3

filipsPL, 2018-10-17:
I will update the scripts. Alternatively you can create a pull request with your version and I will merge it.

filipsPL, 2018-10-17:
Thanks for fixing that!

closed 2018-10-17 by filipsPL