filipsPL/autowx2 #35: Tested same record config as NOAA

SA7BNT, 2018-10-29:
Hey,
i should recommend to use the same recording config (noaa.sh) for ISS record.
Tested in the morning and was a working great.
[20181029-0542_ISS-SSTV.mp3.zip](https://github.com/filipsPL/autowx2/files/2524280/20181029-0542_ISS-SSTV.mp3.zip)
[image]

SA7BNT, 2018-10-29:
I was just running the noaa.sh instead of the iss_voice.sh in the autowx2_config.py
See pic below
I did not convert to MP3. Took the wav from the noaa rec. map and run it in Multiscan 3B SSTV
https://www.qsl.net/kd6cji/
Was working absolutely great for me.

[image]

[image]
Unfortunately I live too far in the north and did not have a good elevation angle.
Here is one more ok pic. with ca 20 degree elevation. 
[image]

filipsPL, 2018-10-30:
Ok, so then it depends on what you are trying to achieve. For recording of voice transmission, I would stay with mp3. For further decoding, like SSTV, wav is better. I will try to modify `iss.sh` to give these two options.

filipsPL, 2018-10-31:
Fixed in 9abe2d9. Now there is an option in `iss_voice.sh` to select mp3 or wav destination. Please test! :wrench:

closed 2018-10-31 by filipsPL

filipsPL, 2018-11-04:
I have decided to do it another way: there are two separate modules:
- for recording wav file
- for recording mp3

and it is configured on the level of `autowx2_conf.py`.

changed in d22d4bc