filipsPL/autowx2 #37: iss_voice.sh in wav mode creating issue

SA7BNT, 2018-11-01:
Just testing to record FOX1D in wav mode, but running in at some issue.
[image]

filipsPL, 2018-11-04:
I have decided to do it another way: there are two separate modules:

- for recording wav file
- for recording mp3

and it is configured on the level of autowx2_conf.py.

changed in d22d4bc

Looking forward the results of tests!

filipsPL, 2019-01-23:
Does it work or not? Can we close the issue? I suppose yes, but need a confirmation.

SA7BNT, 2019-01-23:
Will let you know shortly
[https://twitter.com/RF2Space/status/1087767842785816576?s=19](url)

SA7BNT, 2019-01-31:
Can be closed.
[image]
[image]
[image]

closed 2019-01-31 by SA7BNT