filipsPL/autowx2 #46: Embedding MLRPT for recording Meteor M2 images

DL4OCH, 2018-11-18:
Is it possible, to embed mlrpt for Meteor M2 images as well ? It would be nice, to have a high quality weather-picture source as well.

Should be easy to implement another module for this case, (except for me)...

Best,
 Wilm

filipsPL, 2018-11-19:
Also, please see issue #38

DL4OCH, 2018-11-19:
> Hi Wilm,
> Yes, it should be! Unfortunately I don't have the hardware to test
> this modules, but maybe some other people would like to give it a try?
> If there is a program for Meteor recording/decoding, which can be run
> from the command line, than building a module from it is super-simple.

Hi Filip,

have a look at : http://5b4az.org
Go to weather imaging & Meteor-M ...

There is a piece of software called mlrpt which is a command-line 
receiver/decoder.

http://www.5b4az.org/pkg/lrpt/mlrpt-1.1.tar.bz2

That one even works on Raspberry-Pi, so should be usefull for everybody.

Best,
  Wilm

filipsPL, 2018-11-24:
Good. I was able to download and compile the decoder (tested for `4.18.0-2-686-pae #1 SMP Debian 4.18.10-2 (2018-11-02) i686 GNU/Linux`). As soon as I get some signal probes I will test it and try to make a module out of it.

But definitely we will need some beta testers then... :sweat_smile:

DL4OCH, 2018-11-24:
Hi Filip,
I appreciate your approach. I'm ready to beta-test with my antenna...

Best, 
 Wilm

filipsPL, 2019-01-06:
I've created an **super experimental** module for meteor-m2 support, living in the `modules/meteor-m2`. Please read our wiki page on this topic: https://github.com/filipsPL/autowx2/wiki/meteor-m2 (disclaimer: very crude notes - feel free to edit, polish it, add more info...)
- I was working on mlrpt-1.1, any later version - I was unable to compile (1.2, 1.3)
- I was not able to test it (no hardware), but during dry runs it works.
- please report all issues here, also please modify instructions on the wiki page

happy testing!

DL4OCH, 2019-01-07:
Tested it since yesterday and firgured out, that mlrpt duration is not seconds, but minutes. Means, wrong documentation of mlrpt. We need to hand over minutes, not seconds.

filipsPL, 2019-01-07:
... I think we should care about duration... I've added support for the duration in the meteor module:
`mlrpt -f $freq -s $startT-$stopT -t $durationMin`
available in the current commit.

SA7BNT, 2019-01-09:
Hallo Filip,

i just run the mlrpt by hand. Here i got the pics after Meteor is transmitting again.

When the meteor.sh inside autowx2 starts it seems that there is something going wrong. 
I don’t know what but i should guess it can be from the autowx2_config.py the option 
skipFirst = 20

It seems that after the first 20 sek. he received on the wrong freq. because mlrpt still is running.
A nice feature for mlrpt should be that it is possible to deactivate skipFirst / skipLast option in autowx2_config.py.

Best Tim

Pic´s when i run by hand.

[image]
[image]
[image]
[image]

filipsPL, 2019-01-10:
Congratulations on nice images!

So what if in autowx2_config.py you set: `skipFirst = 0`?

SA7BNT, 2019-01-10:
Will give it a try. Next pass tomorow.

SA7BNT, 2019-01-11:
God morning, tryed to set SkipFirst = 0 and SkipLast = 0 
But i get the same result. 

If you looking at the last part of the log the used freq. is 
> mlrpt: Set SDR Frequency to 461046.5kHz
but why?? Where is he getting the freq. from??

> 2019-01-11 09:12  !! Recording â—  METEOR-M2 :: 2019-01-11 09:11 to 2019-01-11 09:27, dur: 15:30, max el. 69Â°; azimuth: 104Â° (Eâ†’) f=137900000Hz; Decoding: modules/meteor-m2/meteor.sh  
/home/sa7bnt/autowx2/
Bruzaholm
fileNameCore=20190111-0911_METEOR-M2
satellite=METEOR-M2
start=1547194301
duration=911
peak=69
azimuth=104
freq=137900000
0813-0827, duration: 16 min
Found Rafael Micro R820T tuner
Allocating 15 zero-copy buffers
mlrpt: Paused till 08:13 UTC. Operation Timer set to 840 sec
mlrpt: Pause Timer Expired
mlrpt: Operation Timer set to 960 sec
mlrpt: SDR Receiver Type: RTL-SDR
mlrpt: SDR Device Index: 0
mlrpt: SDR Receiver I/Q Sample Rate: 288000
mlrpt: Roofing Filter Bandwidth: 150000
mlrpt: Manual Gain Setting: 90
mlrpt: Frequency Correction Factor: 11
mlrpt: Satellite Frequency kHz: 137900
mlrpt: Decoding Duration: 900
mlrpt: RRC Filter Order: 32
mlrpt: RRC Filter alpha factor: 0.6
mlrpt: Costas PLL Loop Bandwidth: 200
mlrpt: Transmitter QPSK Symbol Rate: 72000
mlrpt: Demodulator Interpolation Factor: 4
mlrpt: LRPT Decoder Output Mode: 3
mlrpt: Save As image file type: 1
mlrpt: JPEG Quality Factor: 85
mlrpt: LRPT Decoder Save Raw Images flag: 0
mlrpt: LRPT Decoder Image Normalize flag: 1
mlrpt: LRPT Decoder Image CLAHE Enhance flag: 1
mlrpt: LRPT Decoder Image Colorize flag: 1
mlrpt: LRPT Decoder Red APID: 68
mlrpt: LRPT Decoder Green APID: 65
mlrpt: LRPT Decoder Blue APID: 64
mlrpt: Invert Palette APID: 68
mlrpt: Red Channel Normalization Range: 0-230
mlrpt: Green Channel Normalization Range: 0-255
mlrpt: Blue Channel Normalization Range: 60-255
mlrpt: Blue Channel min pixel value in pseudo-color image: 60
mlrpt: Blue Channel max pixel value to enhance in pseudo-color image: 80
mlrpt: Blue Channel cloud area pixel value threshold: 210
mlrpt: Directory to save images in: /home/sa7bnt/mlrpt/images/
mlrpt: Initializing RTLSDR Device 
mlrpt: RTLSDR Device Information:
mlrpt: Device Index: 0 Name: Generic RTL2832U OEM
Manufacturer: NOAA Product: RTL2838UHIDIR Serial: 00000002
mlrpt: Tuner Type: R820T
mlrpt: Set SDR Frequency to 461046.5kHz
mlrpt: Frequency Correction: 11 ppm
mlrpt: Setting Tuner Gain Mode to Manual
mlrpt: Setting Tuner Gain to 44dB
mlrpt: ADC Sample Rate: 288000 S/s
mlrpt: RTLSDR Device Initialized OK
mlrpt: Decoding from RTL-SDR Receiver
mlrpt: Operation Timer Started: 960 sec
mlrpt: Decoding of LRPT Images Started
mlrpt: PLL Unlocked
mlrpt: Operation Timer Expired
mlrpt: Receiving and Decoding Ended

DL4OCH, 2019-01-11:
Good morning at all!

Solved the issue by removing the "-f" option in meteor.sh. As Meteor-M2 
only uses 137900kHz, mlrpt defaults will do the job:

http://lampensalat.de:4448/autowx2/mlrpt/images/

Modification in Meteor.sh

mlrpt -s $startT-$stopT -t $durationMin

Best,

  Wilm, DL4OCH

SA7BNT, 2019-01-11:
Well done