filipsPL/autowx2 #52: sox FAIL sox: Rate value `-es' is not a positive number in iss_voice_wav.sh

SA7BNT, 2018-12-04:
Was testing the iss wave script and was running in this issue.
Something you know about??

Tim

2018-12-04 16:16  !! Recording â—        ISS :: 2018-12-04 16:16 to 2018-12-04 16:25, dur: 09:14, max el. 22Â°; azimuth: 163Â° (SSEâ†“â†˜) f=145800000Hz; Decoding: modules/iss/iss_voice_wav.sh  
/home/sa7bnt/autowx2/
Bruzaholm
fileNameCore=20181204-1616_ISS
satellite=ISS
start=1543936588
duration=543
peak=22
azimuth=163
freq=145800000
Recording to wav
Found 2 device(s):
  0:  NOAA, RTL2838UHIDIR, SN: 00000002
  1:  Pocsag, RTL2838UHIDIR, SN: 00000003

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Tuner error set to 11 ppm.
sox FAIL sox: Rate value `-es' is not a positive number
touch: failed to get attributes of '/home/sa7bnt/autowx2/var/www/recordings//iss/rec/2018/12//20181204-1616_ISS.raw': No such file or directory
rm: cannot remove '/home/sa7bnt/autowx2/var/www/recordings//iss/rec/2018/12//20181204-1616_ISS.raw': No such file or directory

filipsPL, 2018-12-11:
Is `iss_voice_wav.sh` created by you or is it a file from the oryginal repository?

SA7BNT, 2018-12-11:
> Is `iss_voice_wav.sh` created by you or is it a file from the oryginal repository?

The file is from the org. repository.
I add this one.
https://github.com/filipsPL/autowx2/commit/a6d3d63cdbb5e7c12ffcba84bececf346d3d449a
And you maked that
https://github.com/filipsPL/autowx2/commit/d22d4bc7b0b2fc7712e59f95408862b4fb513c76
;)

filipsPL, 2018-12-12:
Well... I don't remember why we have `-es` here. Maybe you could record the raw audio file (and don't remove it (just comment the last line `rm $recdir/$fileNameCore.raw`)) and test various options of `sox`? I think this is the only possibility now to debug the problem and test new settings...

SA7BNT, 2018-12-13:
After a little investigate it seems that i got the problem. 
It seems that $sample in the iss_voice_wav.sh is not becomming some value. 
So after i add   
sample="44000" 
in # fixed values for tests just for investigate it seems that it runs without issues.
So the problem seems to be no sample value.

Best 
Tim

SA7BNT, 2018-12-14:
It seems to work.
Here a clip from todays ISS contact.
http://www.ariss-f.org/contact-radioamateur-du-14-12-18/

[ISS_Rec_14-12-2018_1359.mp3.zip](https://github.com/filipsPL/autowx2/files/2680310/ISS_Rec_14-12-2018_1359.mp3.zip)

Best 
Tim

filipsPL, 2018-12-14:
Good job! :+1:

Could you please push changes to the main branch?

closed 2018-12-16 by SA7BNT