filipsPL/autowx2 #58: Meteor-M2 gallary

SA7BNT, 2019-01-11:
After at DL4OCH fix the latest issue with Meteor-M2 and it seems to run stabile, a gallary for Meteor-M2 like NOAA should be nice. :)

https://sa7bnt.funkerportal.de/autowx2//recordings/meteor/img/2019/01/11/

OK1SLM, 2019-01-16:
You have it nice. Somehow I can not set up a beginner.

SA7BNT, 2019-01-16:
> You have it nice. Somehow I can not set up a beginner.

What is the problem?

OK1SLM, 2019-01-16:
modules/meteor-m2/meteor.sh: řádek 50: bc: command not found
5-1816, duration:  min
mlrpt: option requires an argument -- 't'
Usage:  mlrpt -[f<frequency> s<hhmm-hhmm> t<sec> -qihv]
       -f: Specify SDR Receiver Frequency (in kHz).
       -s: Start and Stop Time in hhmm of Operation.
       -t: Duration in min of Operation.
       -q: Run in Quite mode (no messages printed).
       -i: Invert (flip) Images. Useful for South to North passes.
       -h: Print this usage information and exit.
       -v: Print version number and exit.
mlrpt: Paused till 18:15 UTC. Operation Timer set to 60 sec
mlrpt: Pause Timer Expired

SA7BNT, 2019-01-16:
Try to install bc
What system are you using?

OK1SLM, 2019-01-16:
Debian 9   RPI

SA7BNT, 2019-01-16:
try 
sudo apt-get update
sudo apt-get install bc

OK1SLM, 2019-01-16:
Thanks for waiting for another Meteor's flight.

OK1SLM, 2019-01-16:
/root/mlrpt/mlrptrc: No such file or directory
!! mlrpt: Failed to read mlrptrc configuration file - exiting
: Adresář nebo soubor neexistuje

SA7BNT, 2019-01-16:
What are you doing? 
Why run autowx2 as root??

SA7BNT, 2019-01-16:
Also a good idee, take a look into the Wiki
https://github.com/filipsPL/autowx2/wiki/meteor-m2

OK1SLM, 2019-01-16:
I'm sorry to really learn the march.

OK1SLM, 2019-01-16:
Thank you for your patience.

filipsPL, 2019-01-17:
> sudo apt-get update
> sudo apt-get install bc

true, we have to add it to our installation script.

BTW: the gallery for meteor is on the way. Stay tuned!

OK1SLM, 2019-01-17:
Thank you for your help.
Will this change work?

where to place the www data (static html pages)
wwwDir=baseDir + 'var/www/'
Change to
wwwDir='/var/www/html/'

OK1SLM, 2019-01-17:
I still have one question.
How do I connect Main Page to a web server running on the same device?

SA7BNT, 2019-01-17:
https://github.com/filipsPL/autowx2/blob/ae78138764d4a10b7b5f70879a57617bbcb03c9d/autowx2_conf.py.example#L125

OK1SLM, 2019-01-17:
Is there a need to copy something to the new folder?

SA7BNT, 2019-01-17:
No not realy. Just run https://github.com/filipsPL/autowx2/blob/master/bin/gen-static-page.sh

OK1SLM, 2019-01-17:
wwwRootPath='file:///var/www/html/'
I entered the path that I started the script and nothing the html folder is empty.

SA7BNT, 2019-01-17:
wwwRootPath='file:///var/www/html/'
try
wwwRootPath='/autowx2/'
and then "ip/autowx2/index.html" or ip:(webserverport)/autowx2/index.html

OK1SLM, 2019-01-18:
http://www.ok1slm.cz
Meteor-M2

OK1SLM, 2019-01-18:
> wwwRootPath='file:///var/www/html/'
> try
> wwwRootPath='/autowx2/'
> and then "ip/autowx2/index.html" or ip:(webserverport)/autowx2/index.html
It does not work, but I still thank you for your help.

SA7BNT, 2019-01-18:
Then you are doing something wrong. Sorry

OK1SLM, 2019-01-18:
Where does it create a folder and should it create it?

filipsPL, 2019-01-19:
An experimental support for meteor gallery. I can't test it extensively, but please report all bugs here.

[image]

Please remember to refresh/update your config file, and enable meteor logs by:

```
includeGalleryMeteor=1      # Meteor recordings
```
also, please see if modules conf files (like `meteor.conf`) needs to be updated/tuned for operations.

OK1SLM, 2019-01-20:
Which folder should I set up in mlrptrc?

OK1SLM, 2019-01-20:
meteor.sh
52.  source $scriptDir/module_record.sh
to
source $scriptDir/meteor_record.sh

OK1SLM, 2019-01-20:
Satellite: NOAA-19
Pass Start: 2019-01-20 05:29:39 UTC
.Resizing images to 1024 px
**** MCIR-precip
Satellite: NOAA-19
Status: signal processing...........................
Gain: 7.7
Channel A: 3/3B (mid infrared)
convert-im6.q16: invalid argument for option `-resize': 1024x1024\> @ error/convert.c/ConvertImageCommand/2592.
C
Satellite: NOAA-19
Status: signal processing......................mal infrared)
ge
convert-im6.q16: invalid argument for option `-resize': 1024x1024\> @ error/convert.c/ConvertImageCommand/2592Satellite: NOAA-19
Status: signal processing...................
Gain: 7.7
Channel A: 3/3B (mid infrared)
wxtoimg: warning: convert-im6.q16: invalid argument for option `-resize': 1024x1024\> @ error/convert.c/ConvertImageCommand/2592.
**** therm
Satellite: NOAA-19
......Gain: 7.7
Channel A: 3/3B (mid infrared)
t.c/ConvertImageCommand/2592.
**** HVCT-precip
Satellite: NOAA-19
...........wxtoimg: warning: convert-im6.q16: invalid argument for option `-resize': 1024x1024\> @ error/convert.c/ConvertImageCommand/2592.
Satellite: NOAA-19
......Gain: 7.7
Channel B: 4 (thermal infrared)
.
du: nelze přistoupit k '20190120-0630_NOAA-19-MCIR-precip+map.jpg': Adresář nebo soubor neexistuje
: unable to open image `20190120-0630_NOAA-19-MCIR-precip+map.jpg': Adresář nebo soubor neexistuje @ error/blob.c/OpenBlob/2701.
convert-im6.q16: no such image `GRANITE' @ error/mogrify.c/MogrifyImageList/8770.
convert-im6.q16: no images defined `20190120-0630_NOAA-19-MCIR-precip+map.th.jpg' @ error/convert.c/ConvertImageCommand/3258.
du: nelze přistoupit k '20190120-0630_NOAA-19-HVC+map.jpg': Adresář nebo soubor neexistuje
h image `GRANITE' @ error/mogrify.c/MogrifyImageList/8770.
convert-im6.q16: no images defined `20190120-0630_NOAA-19-HVC+map.th.jpg' @ error/convert.c/ConvertImageCommand/3258.

convert-im6.q16: unable to open image `20190120-0630_NOAA-19-MSA+map.jpg': Adresář nebo soubor neexistuje @ error/blob.c/OpenBlob/2701convert-im6.q16: no such image `GRANITE' @ error/mogrify.c/MogrifyImageList/8770convert-im6.q16: no images defined `20190120-0630_NOAA-19-MSA+map.th.jpg' @ error/convert.c/ConvertImageCommand/3258ř nebo soubor neexistuje
/8770.
convert-im6.q16: no images defined `20190120-0630_NOAA-19-therm+map.th.jpg' @ error/convert.c/ConvertImageCommand/3258.

convert-im6.q16: unable to open image `20190120-0630_NOAA-19-HVCT-precip+map.jpg': Adresář nebo soubor neexistuje @ error/blob.c/OpenBlob/2701convert-im6.q16: no such image `GRANITE' @ error/mogrify.c/MogrifyImageList/8770convert-im6.q16: no images defined `20190120-0630_NOAA-19-HVCT-precip+map.th.jpg' @ error/convert.c/ConvertImageCommand/3258convert-im6.q16: unable to open image `20190120-0630_NOAA-19-NO+map.jpg': Adresář nebo soubor neexistuje @ error/blob.c/OpenBlob/2701.
convert-im6.q16: no images defined `20190120-0630_NOAA-19-NO+map.th.jpg' @ error/convert.c/ConvertImageCommand/3258/home/pi/autowx2/

filipsPL, 2019-01-20:
Yes, if you have problems, please comment out in noaa.conf the line to:

```
# resizeimageto=1024
```

OK1SLM, 2019-01-20:
OK Thx

OK1SLM, 2019-01-20:
sed: /home/pi/mlrpt/images//20190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
0190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
sed: /home/pi/mlrpt/images//20190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
sed: /home/pi/mlrpt/images//20190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
date: date: chybné datum:

OK1SLM, 2019-01-20:
sed: /home/pi/mlrpt/images//20190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
0190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
sed: /home/pi/mlrpt/images//20190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje
sed: /home/pi/mlrpt/images//20190120-0935_METEOR-M2.log nelze číst: Adresář nebo soubor neexistuje

OK1SLM, 2019-01-20:
meteor.conf
rawImageDir="/home/pi/mlrpt/images/"

mlrptrc
# Directory to save received images in. Normally it would be
# /home/<user>/mlrpt/images/ if Default is entered here. Else
# the specified directory will be used. Maximum 80 characters.
/home/pi/mlrpt/images/
#/var/lrpt/images/

Is this setting correct?

OK1SLM, 2019-01-20:
[image]

Files have different times.

commit 99620230 (2019-01-20): resize fix #58 #61
closed 2019-01-23 by filipsPL