filipsPL/autowx2 #63: APRS Demodulator does not start

OK1SLM, 2019-01-22:
2019-01-22 16:32  We have still 67:23s free time to the next pass. Let's do something useful!  
2019-01-22 16:32  Running: /home/pi/autowx2/bin/aprs.sh for 67:23s  
/home/pi/autowx2/
Desna
fatal: No names found, cannot describe anything.
Found 2 device(s):
 POCSAG512 POCSAG1200 POCSAG2400 EAS UFSK1200 CLIPFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 STQC DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV SCOPE
multimon-ng  (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
             (C) 2012/2013 by Elias Oenal
             STQC support (C) 2013 Jacek Lipkowski
available demodulators: POCSAG512 POCSAG1200 POCSAG2400 EAS UFSK1200 CLIPFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 STQC DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV SCOPE
  0:  Generic, RTL2832U, SN: 77771111153705700
  1:  Generic, RTL2832U, SN: 77771111153705700

Using device 1: Generic RTL2832U
Warning: -o is very buggy
Found Rafael Micro R828D tuner
Tuner gain set to 49.60 dB.
Tuned to 145064600 Hz.
Oversampling input by: 12x.
Oversampling output by: 4x.
Buffer size: 7.74ms
Exact sample rate is: 1058400.010094 Hz
Sampling at 1058400 S/s.
Output at 22050 Hz.

Please help aprs.sh start but it does not detect anything. After connecting the rtl-sdr receiver to the PC, it all works.

OK1SLM, 2019-01-22:
Tested with one receiver and not working.

SA7BNT, 2019-01-22:
Don´t know what you mean.
First you are writing it works after you replug, then you are writing tested with one stick and it doesen´t work.
Sorry but i can´t help you.

OK1SLM, 2019-01-22:
Does it decode a message for you?

SA7BNT, 2019-01-22:
Don´t know running Radiosonde.

OK1SLM, 2019-01-22:
aprs.sh no run_radiosonde_scanner.sh

filipsPL, 2019-01-22:
Try running from the command line:
```
rtl_fm  -f 144800000 -s 22050 -o 4 -p 0 -g 49.6 | multimon-ng -a AFSK1200 -A -t raw -
```
and see what will happen.

OK1SLM, 2019-01-23:
Install Raspbian 
Install autowx2  error 

Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/cli/base_command.py", line 176, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/commands/install.py", line 346, in run
    session=session, autobuilding=True
  File "/usr/local/lib/python2.7/dist-packages/pip/_internal/wheel.py", line 848, in build
    assert building_is_possible
AssertionError

OK1SLM, 2019-01-23:
SD card  : berryboot
HDD : Raspbian Střech 
Can not it be a problem that Raspbian is not on the card?

OK1SLM, 2019-01-23:
[image]
Real-time PC vs RPI tracking on one antenna

OK1SLM, 2019-01-23:
Resolved by changing the antenna, APRS works. THX 73

OK1SLM, 2019-01-28:
My website running autowx2 : www.ok1slm.cz

closed 2019-02-02 by filipsPL