filipsPL/autowx2 #64: Dongle shift is calculated but not always used

filipsPL, 2019-01-22:
```
2019-01-22 20:40  SDR device found!  
2019-01-22 20:40  Recalibrating the dongle...  
-0.179
2019-01-22 20:40  Using the good old dongle shift (a): 0 ppm 
```

somkar1, 2019-05-09:
For me also dongle shift is not working. It is not even finding new dongle shift and later when I checked in detail, the kal is not finding any frequencies many times.  I tried on debian and also on raspberry 3 b+ both same thing happening.

filipsPL, 2019-05-09:
and what is the output of the `kal` itself? Does it work for you?

somkar1, 2019-05-10:
There are two problems. One is Kal itself doesn't work at times which only its author can solve.

Our problem is calibrate.sh is only run and not the calibrate_full.sh. So there is no dongleshift.txt or gsm_channel.txt files to start with.

channel=$(cat $channelFile) is empty as in autowx2 config file only calibrate is run and calibrate_full is commented by default.

Kal doesn't find anything if the -c $channel is empty. Hence calibrate.sh fails to find the shift.

When manually run output is
-------------------------
$ kal -s GSM900 -e 74 -g 49.6 2 
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
Setting gain: 49.6 dB
kal: Scanning for GSM-900 base stations.
GSM-900:
    chan:   25 (940.0MHz + 733Hz)    power: 4392752.42
    chan:   27 (940.4MHz + 797Hz)    power: 3285022.45
    chan:   29 (940.8MHz + 824Hz)    power: 2098686.73
    chan:   30 (941.0MHz + 1.355kHz)    power: 2575591.98
    chan:   66 (948.2MHz + 145Hz)    power: 1585427.24
    chan:   69 (948.8MHz + 956Hz)    power: 2206713.86
    chan:   70 (949.0MHz + 971Hz)    power: 4093341.39
    chan:   77 (950.4MHz + 1.527kHz)    power: 1795610.18
    chan:   78 (950.6MHz + 1.143kHz)    power: 2098148.96
    chan:   81 (951.2MHz + 1.701kHz)    power: 4590153.19
...chan 124
Difference of offsets between channels is >1kHz. This likely means that the correct PPM is too far away and you need to provide a rough estimate using the '-e' option. Try tuning against a local FM radio or other known frequency first.
--------------------

So this is what calibrate_full will get if we remove the comment. But calibrate_full won't have entered anything on to dongshift.txt($shiftfile) anyway so even if we run calibrate_full.sh still it won't work.

timeout 100 kal -c 27 -g 49.6 -e 0 2
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 270833.002142 Hz
[R82XX] PLL not locked!
Setting gain: 49.6 dB
kal: Calculating clock frequency offset.
Using GSM-900 channel 27 (940.4MHz)
Tuned to 940.400000MHz (reported tuner error: 0Hz)

Hence calibrate.sh fails as there is no value given for -e $recentShift  (which will be zero).