filipsPL/autowx2 #65: Meteor-Module: Add -i parameter for passes

DL4OCH, 2019-01-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

OK1SLM, 2019-01-29:
How to enlarge pictures at Meteor-M2 is it somewhere in mlrptrc?

DL4OCH, 2019-01-29:
No,

at /modules/meteor there is a meteor.conf file. Just a # in front of 
"resizeimageto=1024" (last line) and you're done with full resolution again.

OK1SLM, 2019-01-29:
Yes, thank you, I've already come to that.

filipsPL, 2019-02-07:
How do we know if for the given pass `-i` is needed or not?

OK1SLM, 2019-02-11:
mlrptrc

# Synthesizer Frequency Error correction factor. This is in
# ppm (parts per million) but it has to be an integer number.
# Maximum +/- value for sanity checks is 100.
+49
or
# Synthesizer Frequency Error correction factor. This is in
# ppm (parts per million) but it has to be an integer number.
# Maximum +/- value for sanity checks is 100.
49 
?

SA7BNT, 2019-02-11:
49

OK1SLM, 2019-02-11:
it stopped me from doing meteor-m2 images
www.ok1slm.cz

DL4OCH, 2019-02-11:
Ahoj Libor,

are you sure, that you have 49ppm Offset of your SDR ? Please 
double-check with a local radio-station, as they are quite accurate.

My good old Terratec SDR has an offset of 8 ppm at the requested 
frequency. 49 seems to be way to far.

Best Wishes, Pozdravy,
  Wilm, DL4OCH

filipsPL, 2019-02-12:
please use `bin/calibrate_full.sh` script - it will give you a quite accurate value of your offset.

OK1SLM, 2019-02-12:
> použijte prosím `bin/calibrate_full.sh`skript - dá vám poměrně přesnou hodnotu vašeho offsetu.

one small problem my antenna preamplifier has a GSM band filter

OK1SLM, 2019-02-12:
> Ahoj Libor, are you sure, that you have 49ppm Offset of your SDR ? Please double-check with a local radio-station, as they are quite accurate. My good old Terratec SDR has an offset of 8 ppm at the requested frequency. 49 seems to be way to far. Best Wishes, Pozdravy,  Wilm, DL4OCH
> […](#)

Hi,
images from NOAA are doing fine

filipsPL, 2019-02-12:
> one small problem my antenna preamplifier has a GSM band filter

so you may want to remove the band filter (if possible) and the use `kalibrate` for some time (like 10-20 measurements over 1-2 hours) and take the average of the offset.

OK1SLM, 2019-02-13:
Reinstall RPI2:
Debian 9
RTL-SDR
Autowx2
mrlpt -1.4
Change RTL receiver: 
 RTL2832SDR  R828D TCXO 0,5 PPM 

still black pictures or no pictures
Look at the site there is a log www.ok1slm.cz

DL4OCH, 2019-02-13:
Your setup will not work.
Even a RPi 3B+ will not work with mlrp, as the computing power is far 
too low.
If your going to decode in 3 steps:

1. record a wav-file
2. demodulate it to a soft-QPSK file
3. decode it

It will take on a RPI3 approx. 1 hour.

Best,
  Wilm, DL4OCH

OK1SLM, 2019-02-16:
> Vaše nastavení nebude fungovat. Dokonce i RPi 3B + nebude pracovat s mlrp, protože výpočetní výkon je příliš nízký. Pokud se chystáte dekódovat ve třech krocích: 1. zaznamenat wav soubor 2. demodulovat do soft-QPSK souboru 3. dekódovat to bude trvat cca RPI3. 1 hodina. Nejlepší, Wilm, DL4OCH
> [...](#)

And some idea?
Now I put it on RPI3 B for the future RPI3 B +