filipsPL/autowx2 #66: Twitter integration

Tom-Neverwinter, 2019-02-07:
**Is your feature request related to a problem? Please describe.**
no
**Describe the solution you'd like**
can a auto upload to twitter feature be added :)

**Describe alternatives you've considered**
upload to other social media
**Additional context**
take noaa/goes imagry and upload automatically to social media, like twitter.

filipsPL, 2019-02-07:
We just have to create script which takes a given file(s) and text and sends it to Twitter. I will add it to `noaa.sh` or `meteor.sh`

Some tips:

https://github.com/bear/python-twitter
http://www.tweepy.org/

Tom-Neverwinter, 2019-02-09:
```
import tweepy
from time import sleep

folderpath = "/path/to/your/directory/"
def tweepy_creds():
    consumer_key = 'Removed'
    consumer_secret = 'Removed'
    access_token = 'Removed'
    access_token_secret = 'Removed'

    auth = tweepy.OAuthHandler(consumer_key, consumer_secret)
    auth.set_access_token(access_token, access_token_secret)

    return tweepy.API(auth)

def getPathsFromDir(dir, EXTS="extensions=,png,jpg,jpeg,gif,tif,tiff,tga,bmp"):
    return this.listPaths(folder, EXTS)

def tweet_photos(api):
    imagePaths = getPathsFromDir(this.dataPath(folderpath))
    for x in imagePaths:
        status = "tweet text here"
        try:
            api.update_with_media(filename=x,status=status)
            print "Tweeted!"
            sleep(900)
        except Exception as e:
            print "encountered error! error deets: %s"%str(e)
            break

if __name__ == "__main__":
    tweet_photos(tweepy_creds())
```
code courtesy of: https://stackoverflow.com/questions/51342116/tweepy-auto-tweet-images-from-folder

filipsPL, 2019-02-09:
Thanks @Tom-Neverwinter. Now we need to test it. In my free time I can prepare a beta version of the module, but I need someone to test it later.

SA7BNT, 2019-02-09:
> Thanks @Tom-Neverwinter. Now we need to test it. In my free time I can prepare a beta version of the module, but I need someone to test it later.

You know my mail ;)