filipsPL/autowx2 #70: Radiosonde

OK1SLM, 2019-02-21:
Do you need to install python-eventlet?

filipsPL, 2019-02-21:
Please refer to the [radiosonde manual](https://github.com/projecthorus/radiosonde_auto_rx/wiki#3-setup--configuration) - installation and configuration section. I think `python-eventlet` is not required.

OK1SLM, 2019-02-23:
Yes, I read and I ask.
But another problem after the end of the program is that the program does not end and remains stuck.
I had to restart.

2019-02-23 02:18:01,173 INFO:Habitat - Uploaded sentence to Habitat successfully: $$RS_P3030506,10936,01:18:06,50.65019,14.56482,12941,36.2,-68.5,-1.0,RS41 P3030506 405.100 MHz*15D9
2019-02-23 02:18:30,175 INFO:APRS-IS - Uploaded to APRS-IS: OK1SLM-15>APRARX,SONDEGATE,TCPIP,qAR,OK1SLM-15:;P3030506 *011844h5038.32N/01433.31EO211/076/A=042025 Clb=-3.7m/s t=-67.9C 405.100 MHz Type=RS41 Radiosonde http://bit.ly/2Bj4Sfk !w]{!

2019-02-23 02:19:00,120 INFO:APRS-IS - Uploaded to APRS-IS: OK1SLM-15>APRARX,SONDEGATE,TCPIP,qAR,OK1SLM-15:;P3030506 *011914h5037.72N/01432.89EO200/077/A=041708 Clb=-3.7m/s t=-68.2C 405.100 MHz Type=RS41 Radiosonde http://bit.ly/2Bj4Sfk !w#/!

2019-02-23 02:19:01,064 INFO:Habitat - Uploaded sentence to Habitat successfully: $$RS_P3030506,11004,01:19:14,50.62867,14.54819,12712,40.0,-68.2,-1.0,RS41 P3030506 405.100 MHz*EDDF
2019-02-23 02:19:30,110 ERROR:APRS-IS - Error converting telemetry to APRS packet - 'ascii' codec can't decode byte 0x83 in position 3: ordinal not in range(128)
2019-02-23 02:20:00,147 INFO:APRS-IS - Uploaded to APRS-IS: OK1SLM-15>APRARX,SONDEGATE,TCPIP,qAR,OK1SLM-15:;P3030506 *012013h5036.37N/01432.21EO202/094/A=041110 Clb=-3.5m/s t=-71.4C 405.100 MHz Type=RS41 Radiosonde http://bit.ly/2Bj4Sfk !w#}!

2019-02-23 10:54  âšˇ Program start  
2019-02-23 10:54  Killing all remaining rtl_* processes...  
2019-02-23 10:54  Reading the default dongle shift  
2019-02-23 10:54  Using the default dongle shift: 0 ppm  
2019-02-23 10:54   1. discard NOAA-18, keep ISS  
2019-02-23 10:54  Recalculating the new pass table and saving to disk.  
2019-02-23 10:54   1. discard NOAA-18, keep ISS  
2019-02-23 10:54   2. discard NOAA-19, keep Radiosonde  
2019-02-23 10:54   1. discard NOAA-15, keep ISS  
2019-02-23 10:54   1. discard NOAA-15, keep ISS  
2019-02-23 10:54   1. discard NOAA-18, keep ISS  
2019-02-23 10:54   1. discard NOAA-18, keep ISS

OK1SLM, 2019-02-24:
pi@raspberrypi:~ $ cd autowx2
pi@raspberrypi:~/autowx2 $ python autowx2.py
2019-02-24 12:48  ⚡ Program start  
2019-02-24 12:48  Killing all remaining rtl_* processes...  
WebSocket transport not available. Install eventlet or gevent and gevent-websocket for improved performance.
2019-02-24 12:48  Reading the default dongle shift 

?????

filipsPL, 2019-02-27:
So try to install both of them:
```
pip install eventlet
pip install gevent-websocket
```
and see.

OK1SLM, 2019-02-27:
radiosonde_auto_rx

Note: If you have either gevent or eventlet installed, these will need to be uninstalled so that the Flask-SocketIO does not try to use it. If you didn't understand that sentence then you're probably OK. If you did understand it, then you'll need to uninstall gevent and/or eventlet until we figure out a workaround. Sorry :-(

OK1SLM, 2019-05-12:
Any idea?