filipsPL/autowx2 #71: Webserver on rpi3

Blackbeard25, 2019-03-13:
I am unable to access the webserver or view the images recorded. using a rpi 3B running raspbian lite

filipsPL, 2019-03-13:
- Any errors or warnings when starting autowx2? Any error or warnings when running and trying to access the webserver?
- Did you configure the webserver port in the config file?
- When trying to access the webserver, whar is the result (blank page, error?
- are you trying to access it from a local machine or a remote location?

Blackbeard25, 2019-03-13:
I am trying to access it from a machine on the local network, no errors appear, I left the port number the same, the error was connection refused

SA7BNT, 2019-03-13:
Hey @Blackbeard25 

please share your autowx2_conf.py with us. 

The images will be saved on depence what you have set in the config file.
Take a look here
[https://github.com/filipsPL/autowx2/blob/1ceb5fd1ac3e86fc0f5ebca7c995866a2060fe1c/autowx2_conf.py.example#L108](url)

Blackbeard25, 2019-03-13:
@SA7BNT 
Here is my autowx_conf.py. I got a image output but but no image, just the map overlay. And there are no wav files where there are supposed to be 
#
# autowx2 - config file
# variables used by autowx2
#

#
# baseDir is taken from this script
# do not change, run configure.sh to configure
#
from basedir_conf import *

# satellites to record
# for list of active satellites and used frequences, see
# http://www.dk3wn.info/p/?page_id=29535

satellitesData = {
    'NOAA-18': {
        'freq': '137912500',
        'processWith': 'modules/noaa/noaa.sh',
        'priority': 2},
    'NOAA-15': {
        'freq': '137620000',
        'processWith': 'modules/noaa/noaa.sh',
        'priority': 2},
    'NOAA-19': {
        'freq': '137100000',
        'processWith': 'modules/noaa/noaa.sh',
        'priority': 2},
#    'ISS': {
#        #'freq': '145800000',  # FM U/v VOICE Repeater (Worldwide) and FM SSTV downlink (Worldwide) [OK]
#        'freq': '143625000',  # FM VHF-1 downlink. Main Russian communications channel. Often active over Moskow. [ ]
#        #'freq': '145825000', # APRS -- AX.25 1200 Bd AFSK Packet Radio (Worldwide) Downlink [ ]
#        'processWith': 'modules/iss/iss_voice_mp3.sh',
#        'priority': 1},
#    'Radiosonde': {		# meteo balloons launched twice a day, monitored for two hours
#        'freq': '98796500',
#        'processWith': 'modules/radiosonde/run_radiosonde_scanner.sh',
#        'fixedTime': '20 0,12 * * *',
#        'fixedDuration': 7200,
#        'priority': 1},
    #'FOX-1A': {  # http://www.dk3wn.info/p/?cat=80
        #'freq': '145980000',
        #'processWith': 'modules/iss/iss_voice.sh',
        #'priority': 4},
    #'FOX-1D': {  # http://www.dk3wn.info/p/?cat=80
        #'freq': '145880000',
        #'processWith': 'modules/iss/iss_voice.sh',
        #'priority': 4},
    #'FOX-1B': {  # http://www.dk3wn.info/p/?cat=80
        #'freq': '145960000',
        #'processWith': 'modules/iss/iss_voice.sh',
        #'priority': 4},

}

#
# priority time margin - time margin between two overlapping transits with different priorities, in seconds
#
priorityTimeMargin = 240

#
# minimal elevation of pass to capture satellite
#
minElev = 10

#
# skip first n seconds of the pass
#
skipFirst = 10

#
# skip last n seconds of the pass
#
skipLast = 10

# staion information
# lon: positive values for W, negative for E
stationLat = '52.8409064'
stationLon = '109.828223'
stationAlt = '650'
stationName = 'Marsden Sask'

#
# dongle gain
#
dongleGain = '49.6'

#
# Kill all rtl_ related processes after/berofr recordings?
# This should be enabled when the user doesn't use another instances which may run rtl software
#

cleanupRtl = True

#
# where to place the www data (static html pages)
#
wwwDir=baseDir + 'var/www/'

#
# recording direcotry - where to put recorded files, processed images etc. - the core
#
recordingDir = baseDir + "recordings/"         # will goto recording dir in local dir
#recordingDir = wwwDir + "recordings/"		# will go to www dir

#
# location of the HTML file with a list of next passes
#
htmlNextPassList = wwwDir + 'nextpass.tmp'

#
# location of the Gantt chart file with a plot of next passes; PNG or SVG file extension possible
#
ganttNextPassList = wwwDir + 'nextpass.png'

#
# WWW root path - where is your main index.html file
# this could be:
# / - for page in the root www directory, eg. www.domain.org/index.html
# /~filipsPL/ - for page located in filipsPL's home directory
# file:///home/filipsPL/github/autowx2/var/www/ - for files stored locally in the given local directory

wwwRootPath='/'

#
# galleries to include in the static index page
# 1 = inclcude, 0 = don't include
#
includeGalleryNoaa=1        # NOAA recordings
includeGalleryISS=0         # ISS recordings
includeGalleryMeteor=1      # Meteor recordings
includeGalleryLogs=1        # logs in plain text
includeGalleryDump1090=0    # dump1090 data

#
# port at which webserver will be listening
#
webInterfacePort = 5010

# script tha will be used whle waiting for the next pass; set False if we just want to sleep
# by default, this script will get the parameter of duration of the time to be run and the recent dongleShift
scriptToRunInFreeTime = False				# does nothing
#scriptToRunInFreeTime = baseDir + "bin/aprs.sh" 	# APRS monitor
#scriptToRunInFreeTime = baseDir + "bin/radiosonde_auto_rx.sh" # radiosonde tracker, see https://github.com/projecthorus/radiosonde_auto_rx
#scriptToRunInFreeTime = baseDir + "bin/pymultimonaprs.sh" # APRS iGate,

# pymultimonaprs must be installed, see: https://github.com/asdil12/pymultimonaprs/
#scriptToRunInFreeTime = baseDir + "bin/dump1090.sh"

#scriptToRunInFreeTime = baseDir + "bin/rtl_433.sh" 	# monitor traffic on 433 MHz; see: rtl_433

### Logging to file (enter the path) or False to disable
# loggingDir = False
loggingDir = recordingDir + "/logs/"

# Dongle PPM shift, hopefully this will change to reflect different PPM on freq
dongleShift = '0'

#
# wether to turn bias t power on (i.e., for LNA)
# see: rtl_fm --help | grep "-T"
# biast=""
# biast="-T"
#
biast=""

#
tleDir = baseDir + 'var/tle/'
tleFile = 'all.txt'

tleFileName = tleDir + tleFile

# dongle shift file
dongleShiftFile = baseDir + "var/dongleshift.txt"

# dongle calibration program
# should return the dongle ppm shift

calibrationTool = False # don't calibrate the dongle, use old/fixed shift
#calibrationTool = baseDir + "bin/calibrate.sh"         # uses predefined GSM channel
#calibrationTool = baseDir + "bin/calibrate_full.sh"     # check for the best GSM channel and calibrates

# DERIVATIVES #############################

#
# latlonalt for use with wxmap
#
latlonalt = "%s/%s/%s" % (stationLat, -1 * float(stationLon), stationAlt)

SA7BNT, 2019-03-13:
What is in the logfile? 
To find under /recordings/logs/

What kind of reciver are you using? 
Why not calibrate if it is a RTL DVB-T dongle??

Here is my conf 
[autowx2_conf.txt](https://github.com/filipsPL/autowx2/files/2963495/autowx2_conf.txt)

iw5ejm, 2019-06-02:
same issue here but I have no usr/local/lib/python2.7/dist-packages/flask_socketio/init.py file.

All required python modules are correctly installed.