filipsPL/autowx2 #77: ImportError: No module named predict

Milan12332567, 2019-04-20:
Hello
So whatever script i run, i get this error:

`pi@raspberrypi:~/autowx2 $ python autowx2.py
Traceback (most recent call last):
  File "autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/pi/autowx2/autowx2_functions.py", line 14, in <module>
    import predict
ImportError: No module named predict
`

I used regular installation, can you help me?

OK1SLM, 2019-04-20:
Hi
install sudo bash install.sh
no bash install.sh

Milan12332567, 2019-04-20:
hello, i already got installed sudo; bash

Milan12332567, 2019-04-20:
im sorry i misunderstood

Milan12332567, 2019-04-20:
hello so i got another problem:
im 2 hours now stuck at this line:

`******** Getting auxiliary programs

--2019-04-20 16:48:51--  https://raw.githubusercontent.com/filipsPL/heatmap/master/heatmap.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 
`
i dont really know what to do in this situation

filipsPL, 2019-04-24:
Do you have a stable internet connection? Looks like a problems with fetching some files from the Internet...