filipsPL/autowx2 #78: Getting auxiliary programs stuck (waited 4 hours)

Milan12332567, 2019-04-20:
Hello so I tryed installing this product, but i am stuck here:
--2019-04-20 16:48:51--  https://raw.githubusercontent.com/filipsPL/heatmap/master/heatmap.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... .

its 20:11

can you help me? i think i need assistance with installation

SA7BNT, 2019-04-20:
First of all let us know what system are you using. Desktop PC or a Raspberry Pi ??
What have you done.
And what will you use the heatmap for?? If i´m right there was a smal problem wiyh that.

Milan12332567, 2019-04-20:
Im using Raspberry PI 3 B+ and i just run this command: ` sudo bash install.sh ` , I didnt changed anything in script

SA7BNT, 2019-04-20:
Did you get error messages when you run the install.sh?

Milan12332567, 2019-04-21:
I'm running lastest raspbian lite. No I didn't get any error messages, it just get stuck.

SA7BNT, 2019-04-21:
Ok, i will try to make a run the next days and let you know.

SA7BNT, 2019-04-23:
Just maked a fresh Raspbian Lite in a SD and installed autowx2 via sudo bash install.sh 
No issue all is running fine. 
Also the clone of the [https://github.com/filipsPL/heatmap](url) and python install of osmviz was running without issues. 

Best Tim

Milan12332567, 2019-04-24:
ok thanks
ill try later