filipsPL/autowx2 #79: Deactivate option for Flask web server

SA7BNT, 2019-04-21:
Is it possible in the future to have an option in the autowx2_config.py to deactivate the "Flask web server" if it not will be used?
My problem is that if i run autowx2 in a screen session, and stop that screen session, because i want to configure something, the "Flask web server" sometimes will not stop.

Best
Tim

filipsPL, 2019-04-23:
Good point. I will make flask webserver optional.

SA7BNT, 2020-03-03:
Hey, are you still active on this project?? Is it possible to add this function ?? It seems that flask is eating my ram. Hope to hear from you.
Best

filipsPL, 2020-03-04:
I'm here, but not super-active. I will try to look at this option soon - maybe I will remove the flask web server entirely...