filipsPL/autowx2 #80: recording of noaa does not work

smarti2019, 2019-04-27:
I have a fresh raspian installation. The installation of autowx2 has worked so far. No error message. But there is no noaa recording.

SA7BNT, 2019-04-27:
Please post your log to see where you got problems

OK1SLM, 2019-04-27:
sudo wxtoimg 
:yes
Enter

smarti2019, 2019-04-28:
sudo wxtoimg with yes entered ok, but no record from noaa.
Where is the log file?
Here is an abstract of the last consolen messages:

2019-04-28 10:50   2. discard NOAA-18, keep Radiosonde
2019-04-28 10:50   1. discard NOAA-18, keep Radiosonde
2019-04-28 10:50  Recalculating the new pass table and saving to disk.
2019-04-28 10:50   2. discard NOAA-18, keep Radiosonde
2019-04-28 10:50   1. discard NOAA-18, keep Radiosonde
2019-04-28 10:50   1. discard NOAA-15, keep ISS
2019-04-28 10:50   2. discard NOAA-19, keep ISS
This is a miracle!
2019-04-28 10:50  Next five passes:
2019-04-28 10:50  ●    NOAA-18 :: 2019-04-28 10:48 to 2019-04-28 11:03, dur: 15:15, max el. 81°; azimuth: 108° (E→) f=137912500Hz; Decoding: modules/noaa/noaa.sh
2019-04-28 10:50  ● Radiosonde :: 2019-04-28 12:20 to 2019-04-28 14:20, dur: 120:00, max el. 0°; azimuth: 0° (N↑) f=98796500Hz; Decoding: modules/radiosonde/run_radiosonde_scanner.sh
2019-04-28 10:50  ●    NOAA-19 :: 2019-04-28 15:11 to 2019-04-28 15:24, dur: 13:20, max el. 22°; azimuth: 44° (NNE↑↗) f=137100000Hz; Decoding: modules/noaa/noaa.sh
2019-04-28 10:50  ●    NOAA-19 :: 2019-04-28 16:50 to 2019-04-28 17:05, dur: 15:05, max el. 72°; azimuth: 66° (NE↗) f=137100000Hz; Decoding: modules/noaa/noaa.sh
2019-04-28 10:50  ●    NOAA-19 :: 2019-04-28 18:31 to 2019-04-28 18:46, dur: 14:14, max el. 28°; azimuth: 267° (WSW←↙) f=137100000Hz; Decoding: modules/noaa/noaa.sh
2019-04-28 10:50  Next pass:
2019-04-28 10:50  ●    NOAA-18 :: 2019-04-28 10:48 to 2019-04-28 11:03, dur: 15:15, max el. 81°; azimuth: 108° (E→) f=137912500Hz; Decoding: modules/noaa/noaa.sh
2019-04-28 10:50  Killing all remaining rtl_* processes...
2019-04-28 10:50  !! Recording ●    NOAA-18 :: 2019-04-28 10:48 to 2019-04-28 11:03, dur: 15:15, max el. 81°; azimuth: 108° (E→) f=137912500Hz; Decoding: modules/noaa/noaa.sh
/home/pi/autowx2/
HOME
fileNameCore=20190428-1048_NOAA-18
satellite=NOAA-18
start=1556441288
duration=744
peak=81
azimuth=108
freq=137912500
sample=
wavrate=
dongleIndex=
enchancements=

But i find no files, no wav, no raw, no jpeg,
rtl_433 works during the breaks

SA7BNT, 2019-04-28:
You can try to run the test.
https://github.com/filipsPL/autowx2/blob/72804c588a1d793e2b5f76471a67e55d0d7201ab/modules/noaa/noaa.sh#L46
Remove the # and run 
./noaa.sh tests/20180118-1504_NOAA-19.wav tests/20180118-1504_NOAA-19 NOAA-19 1516284265 926
from the commandline. 
See what will happend.

smarti2019, 2019-04-29:
at the console repeated and repeated:

HOME
fatal: Keine Namen gefunden, kann nichts beschreiben.
fileNameCore=tests/20180118-1504_NOAA-19.wav
satellite=tests/20180118-1504_NOAA-19
start=NOAA-19
duration=1516284265
peak=926
azimuth=
freq=
sample=48000
wavrate=11025
dongleIndex=0
enchancements=MCIR-precip
/home/pi/autowx2/
HOME
fatal: Keine Namen gefunden, kann nichts beschreiben.
fileNameCore=tests/20180118-1504_NOAA-19.wav
satellite=tests/20180118-1504_NOAA-19
start=NOAA-19
duration=1516284265
peak=926
azimuth=
freq=
sample=48000
wavrate=11025
dongleIndex=0
enchancements=MCIR-precip
/home/pi/autowx2/

SA7BNT, 2019-04-30:
Strange!! Maybe @filipsPL have some idee.

smarti2019, 2019-04-30:
I'm sure you can adjust that too.
Just a fresh raspberry pi with rtl-sdr v3 dongel and current raspian.
pulled autowx2 from the github and then run install.sh.
the problem is reproducible

SA7BNT, 2019-04-30:
> sudo wxtoimg with yes entered ok, but no record from noaa.
> Where is the log file?
> Here is an abstract of the last consolen messages:
>
> 2019-04-28 10:50 2. discard NOAA-18, keep Radiosonde
> 2019-04-28 [...]

Is your time (regional settings) right for your place where you are??
Just tested, and i was not running in at some issue.
There was just a RTL-SDR stick connected NO ANTENNA on the stick. I just have one antenna for my running system. 
My running system is NOT a Raspberry Pi. It´s a Dell Optiplex I5 because of Meteor-M2 is taking a lot of CPU performance.
You can see it at [https://sa7bnt.funkerportal.de/autowx2/](url)
Best Tim

[image]

filipsPL, 2019-05-06:
Do you have the `rtl_fm` installed and running? What if you type `rtl_fm` in the console?

And, if it is ok, try to listen to some fm transmission, eg: `rtl_fm -f 89.1M -M fm -s 170k -A fast -r 32k -l 0 -E deemp | play -r 32k`