filipsPL/autowx2 #81: Can't find TLE data (in keplers) nor fixed time schedule (in config)

somkar1, 2019-05-16:
I am running this for the past few days and everything was working fine as intended. But suddenly today this problem has started and what ever pass which were to happen after 5.31 local time (cron was set to update keps at 5.31) were not coming. When I checked this is error  I get.

-------------------------------------
2019-05-16 12:22  ✖ Can't find TLE data (in keplers) nor fixed time schedule (in config) for NOAA-18  
2019-05-16 12:22  ✖ Can't find TLE data (in keplers) nor fixed time schedule (in config) for NOAA-19  
2019-05-16 12:22  ✖ Can't find TLE data (in keplers) nor fixed time schedule (in config) for NOAA-15  
2019-05-16 12:22  Recalculating the new pass table and saving to disk.  
2019-05-16 12:22  ✖ Can't find TLE data (in keplers) nor fixed time schedule (in config) for NOAA-18  
2019-05-16 12:22  ✖ Can't find TLE data (in keplers) nor fixed time schedule (in config) for NOAA-19  
2019-05-16 12:22  ✖ Can't find TLE data (in keplers) nor fixed time schedule (in config) for NOAA-15  
This is a miracle!
----------------------------------

I checked few other sites which use autowx2 are also blank without pass list being mentioned.

filipsPL, 2019-05-16:
Hi, thanks for reporting!

I guess it is because `http://www.pe0sat.vgnet.nl/kepler/mykepler.txt` is empty, and in the remaining tle files NOAA satellites are represented not as `NOAA-15` but as `NOAA 15`.

A temporary workaround would be a change in config file from `NOAA-xx` to `NOAA xx`.

I will try to add a mechanism for dealing with such situaltions.

SA7BNT, 2019-05-16:
Same is on Meteor-M2 
Change NOAA-15, NOAA-18, NOAA-19,Meteor-M2 ---> Meteor-M 2, NOAA 15, NOAA 18, NOAA 19 in autowx2_config.py
Then python genpasstable.py and /bin/gen-static-page.sh

Then all is running fine again.

somkar1, 2019-05-16:
Thanks, I added this to upload-keps.sh in the hope that from now on it will do it automatically. Actually I should have used $TLE variable instead of fixed directory name.
---------------------------
sed -i 's/NOAA 15/NOAA-15/g' /home/pi/autowx2/var/tle/weather.txt
sed -i 's/NOAA 18/NOAA-18/g' /home/pi/autowx2/var/tle/weather.txt
sed -i 's/NOAA 19/NOAA-19/g' /home/pi/autowx2/var/tle/weather.txt
sed -i 's/METEOR-M 2/METEOR-M2/g' /home/pi/autowx2/var/tle/weather.txt

sed -i 's/NOAA 15/NOAA-15/g' /home/pi/autowx2/var/tle/all.txt
sed -i 's/NOAA 18/NOAA-18/g' /home/pi/autowx2/var/tle/all.txt
sed -i 's/NOAA 19/NOAA-19/g' /home/pi/autowx2/var/tle/all.txt
sed -i 's/METEOR-M 2/METEOR-M2/g' /home/pi/autowx2/var/tle/all.txt
---------------------------

filipsPL, 2019-05-16:
Thanks @somkar1 for this hack, I will add it to the repo meanwhile!