filipsPL/autowx2 #83: METEOR-M2 2 Reception

lic2h5o, 2019-07-19:
Add support for the new METEOR-M2 2 satellite (LRPT, 137.9Mhz)( https://www.rtl-sdr.com/weather-satellite-meteor-m2-2-now-transmitting-images/ )

filipsPL, 2019-07-19:
Do we have a linux command line tool for that? If yes, let's try!

SA7BNT, 2019-07-19:
Nope not yet. Talking to Neoklis and he has not finished it.
Will let you know when it´s time

Best Tim

> Tim, 
mlrpt reads the config file very early and reports the parameters it reads, mainly for debugging reasons. The actual receiver frequency, if one is specified on the command line, is reported in the last few lines of mlrpt's printout. 
I agree this is somewhat confusing but it has to stay as its a good idea to see what parameters are in the config file. But now with the new sat I have modified mlrpt to read a config file specified on the command line, like:
mlrpt -c Meteor_M2-2.cfg so that a separate config file for each sat can be made.
I will release this new version later once it is tested and finalized.

SA7BNT, 2019-07-19:
Bad news 

> Tim
Looks like mlrpt/glrpt cannot decode OQPSK and for now I don't know how to make changes. Hopefully the developer of the original code will help.

SA7BNT, 2019-07-19:
> Add support for the new METEOR-M2 2 satellite (LRPT, 137.9Mhz)( https://www.rtl-sdr.com/weather-satellite-meteor-m2-2-now-transmitting-images/ )

You can look here
[https://github.com/artlav/meteor_decoder/issues/7](url)