filipsPL/autowx2 #84: Autostart script

OK1SLM, 2019-07-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

OK1SLM, 2019-07-20:
You can do autostart script on RPI?

0nnyx, 2019-08-22:
This will create a service which will start at bootup and restart if autowx dies (which happens) :

sudo nano /etc/systemd/system autowx2.service
> [Unit]
> Description=autowx2
>[Service]
>Restart=always
>RestartSec=10
>WorkingDirectory=/home/pi/autowx2
>ExecStart=/home/pi/autowx2/autowx2.py
>[Install]
>WantedBy=multi-user.target

_Replace /home/pi with the location of your autowx2 directory_

sudo systemctl enable autowx2.service
sudo systemctl start autowx2.service

Check if working properly with : systemctl status autowx2.service

OK1SLM, 2019-08-22:
sudo nano /lib/systemd/system/autowx2.service

[Unit]
Description=autowx2
After=multi-user.target
[Service]
Restart=always
RestartSec=10
WorkingDirectory=/home/pi/autowx2
ExecStart=/usr/bin/python /home/pi/autowx2/autowx2.py
[Install]
WantedBy=multi-user.target

sudo chmod 644 /lib/systemd/system/autowx2.service
sudo systemctl daemon-reload
sudo systemctl enable autowx2.service
sudo systemctl start autowx2.service

Eric738, 2020-06-06:
gm 
i follow your script but raspi answer after cmd "sudo systemctl enable autowx2.service" : Failed to enable unit: Unit file autowx2.service does not exist. 
can your help Me ?
73 Eric F1SMV

0nnyx, 2020-06-06:
Hi Eric, your error means the autowx2.service file was not created. I just updated my initial post to be fully correct & more explicit. Try it again

Eric738, 2020-06-07:
GA

follow your nex instructions, but always this error message...
--------------------- 
   Loaded: loaded (/etc/systemd/system/autowx2.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-06-07 18:15:35 CEST; 21s ago
 Main PID: 3230 (python)
    Tasks: 2 (limit: 2200)
   Memory: 49.1M
   CGroup: /system.slice/autowx2.service
           └─3230 python /home/pi/autowx2/autowx2.py

juin 07 18:15:51 raspberrypi autowx2.py[3230]:     self.__target(*self.__args, **self.__kwargs)
juin 07 18:15:51 raspberrypi autowx2.py[3230]:   File "/home/pi/autowx2/autowx2_functions.py", line 722, in mainLoop
juin 07 18:15:51 raspberrypi autowx2.py[3230]:     while not runTest():
juin 07 18:15:51 raspberrypi autowx2.py[3230]:   File "/home/pi/autowx2/autowx2_functions.py", line 268, in runTest
juin 07 18:15:51 raspberrypi autowx2.py[3230]:     stderr=subprocess.PIPE)
juin 07 18:15:51 raspberrypi autowx2.py[3230]:   File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
juin 07 18:15:51 raspberrypi autowx2.py[3230]:     errread, errwrite)
juin 07 18:15:51 raspberrypi autowx2.py[3230]:   File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
juin 07 18:15:51 raspberrypi autowx2.py[3230]:     raise child_exception
juin 07 18:15:51 raspberrypi autowx2.py[3230]: OSError: [Errno 2] No such file or directory
------------------------------------------
i updated python for version 3.8.0 

Eric

SA7BNT, 2020-06-07:
Mine is running in a screen session.

[Unit]
Description=AutoWX2

[Service]
RemainAfterExit=yes
User=sa7bnt
ExecStart=/usr/bin/screen -dmS autowx2 /usr/bin/python /home/sa7bnt/autowx2/autowx2.py
ExecStop=/usr/bin/screen -S autowx2 -X quit

[Install]
WantedBy=multi-user.target