filipsPL/autowx2 #86: Meteor_M2_2

SA7BNT, 2019-09-09:
Hallo Filip,

MLRPT http://www.5b4az.org/ is now out in a BETA version. I tested it here and it seems to running „well“ 
The mlrpt in my autowx2 is running with 2 different conf. for mlrpt
One for M2 and one for M2_2
I have copied the Meteor_M2 module for M2_2 and just add the command -c <config-file> to the meteor.sh
Seems to running well. But all Meteor is now in one Meteor gallery. If your time allows it can you please add a separate gallery for Meteor M2_2 in Git Hub autowx2??

Best from Sweden

Tim

OK1SLM, 2019-09-20:
So the first images of autowx Meteor M2, Meteor M2-2
http://www.ok1slm.cz/recordings/meteor/img/2019/09/20/index.html

SA7BNT, 2019-09-20:
Please share your config for mlrpt M2 and M2_2

OK1SLM, 2019-09-20:
Config
https://gofile.io/?c=IOF5fV

SA7BNT, 2019-09-21:
Yea, first great pic M2_2
[image]

ChadOhman, 2020-01-13:
FYI Meteor M2-2 is now deactivated due to a meteor strike and resulting decompression: http://happysat.nl/Meteor/html/Meteor_Status.html