filipsPL/autowx2 #87: `sox` recording failing, and other issues (with solutions)

ThomasBartleet, 2019-11-12:
Hi,

Thank you for providing this great repository! Once it was up and running, it was really good to see images processed and appearing on my RPi!

I did stumble across some issues along the way, which after hours (even days!) of Google/SO searching, I managed to solve. I'm writing this in the hope that someone else experiencing what I had can also try to fix.

**Issue 1:** The first time I had run the `install.sh` script, I noticed in the logs a tonne of warnings and errors in the console logs. 
**Solution:** Literally googled the error messages and there were a number of solutions for each warning/error that appeared. Once I applied each potential fix to each warning/error message, I re-ran the `install.sh` script. **Note**: the installation script should be free of error messages. Warnings saying one library was installed instead of another should be ok, but any other warning messages (for example, regarding the C code) **should not be ignored**. I found sometimes that running the `install.sh` script several times cleared some warning/error messages (maybe libraries were installed in the wrong order??).

**Issue 2:** Once the above issues were addressed and resolved (no error or warning messages in the console), I ran the `autowx2.py`. For the most part, it worked fine, until it came to recording a pass. It so happened that the `sox` command would crash after 10-120 seconds, without saving any recordings, not process anything, and try again during the same pass. This happened for every pass. I added some `echo` and `print()` commands to both the bash and Python scripts to make sure the commands passed had the correct parameters (duration, correct paths, etc). All looked fine, but clearly something was going wrong.
**Solution:** Not knowing what was causing `sox` to fail, I started removing all the unnecessary features for recording a fly-over, the biggest being the webserver. Having no webserver now, I figured there was no need to create a new thread for the satellite stuff. In the `autowx2.py`, I replaced
```python
while True:
    t1 = Thread(target = mainLoop)
    t1.setDaemon(True)
    t1.start()
    # app.run(debug=True, port=webInterfacePort)

    socketio.run(app, port=webInterfacePort, debug=False)
```
with
```python
while True:
    try:
        mainLoop()
    finally:
        print("[MAIN] Main loop exited for some reason. Check the logs.")
```

This allowed the `sox` command to run for the entire pass duration and actually record a pass. From there, the pass was processed, and I could actually see the images generated. Having done the above, I tried re-implementing the web stuff (using `socketio.run(...)`), but `sox` would fail again. Clearly something was (is?) going wrong with implementing the web stuff in Python whilst simultaneously doing the predicting, recording, and processing of a pass.

**Issue 3:** Following on from the above, I still would have liked a web server to view the images. However, as established, using the current script to run a server in Python AND do the satellite handling doesn't work.
**Solution:** I implemented a server using NodeJS and the Express framework. I had to tweak the templates and static file generation methods to accommodate the Express framework instead of Flask, and I now have my RPi running two terminals: one with the satellite processing, and the other with the NodeJS web server. This is now working perfectly.

### A bit more info

- I have an RPi 3 Model B+ set up running the fixes above. 
- I have a TA-1 cross-dipole 135-157MHz, hooked into an RTL-SDR, which is plugged into the RPi.
- This has been running continuously for the last few weeks. The images have been processed and are on display on my webserver, accessible through my browser.
- The RPi has been a bit warm, but seems to be running fine.

I hope the information is helpful to someone out there who experienced similar issues as I have!

filipsPL, 2019-11-12:
@ThomasBartleet Thank you so much for providing tips on installing and runnig autowx2! I will try to implement some ot those directly in the code. I will also move the whole document to our wiki (or - better - if you could create an appropriate page and copy it there).

ThomasBartleet, 2019-11-13:
@filipsPL No worries, I'll have a look at writing something up, but it won't be for another week or so, as I have other commitments going on!