filipsPL/autowx2 #89: rtl_tcp

ChadOhman, 2020-01-12:
I have a quadrafilar helix antenna with a raspberry pi zero in the mast running rtl_tcp.

Is there a way to use rtl_tcp with autowx2?

Thanks!

SA7BNT, 2020-01-12:
> I have a quadrafilar helix antenna with a raspberry pi zero in the mast running rtl_tcp.
> 
> Is there a way to use rtl_tcp with autowx2?
> 
> Thanks!

Hey, you can try to change it here and see how it works.
https://github.com/filipsPL/autowx2/blob/72804c588a1d793e2b5f76471a67e55d0d7201ab/modules/noaa/noaa_record.sh#L11

Best Tim
SA7BNT

ChadOhman, 2020-01-12:
Hi!
Thanks for the reply. What I ended up doing was implementing USBIP, I took portions of this tutorial and the RTLSDR shows up like a physical USB device on the client: https://derushadigital.com/other%20projects/2019/02/19/RPi-USBIP-ZWave.html

closed 2020-01-12 by ChadOhman