filipsPL/autowx2 #90: OSError: [Errno 2] No such file or directory

ChadOhman, 2020-01-12:
Hi!

Got things up and running, however I'm facing this error:

```2020-01-12 23:31  Killing all remaining rtl_* processes...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/chad/autowx2/autowx2_functions.py", line 722, in mainLoop
    while not runTest():
  File "/home/chad/autowx2/autowx2_functions.py", line 268, in runTest
    stderr=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/chad/autowx2/autowx2_functions.py", line 722, in mainLoop
    while not runTest():
  File "/home/chad/autowx2/autowx2_functions.py", line 268, in runTest
    stderr=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

Has anyone seen this before? Is this normal?

SA7BNT, 2020-01-13:
Try to reboot your system. Or change the port from flask before you start autowx2.py
https://github.com/filipsPL/autowx2/blob/72804c588a1d793e2b5f76471a67e55d0d7201ab/autowx2_conf.py.example#L149

ChadOhman, 2020-01-13:
Thanks for your reply, @SA7BNT.

Changing the port didn't work to get the flask server up - however I'm not seeing the error I was seeing before anymore. I think it's listening on localhost so it won't accept traffic on eth0.

SA7BNT, 2020-01-13:
You may try to install lighttpd or apache2. 
Then point the autowx2 config to your page
https://github.com/filipsPL/autowx2/blob/72804c588a1d793e2b5f76471a67e55d0d7201ab/autowx2_conf.py.example#L102
Maybe that will help you

ChadOhman, 2020-01-13:
Thanks. Installed apache2, pointed autowx2 to /var/www/html, and nada.
Manually copied autowx2/var/www/ to the apache2 web root and while that did create content, I've noted all of the referenced elements on the page have an extra / in front of them. I tried stripping out all of them but something doesn't seem right. I don't know if this is the solution.

ChadOhman, 2020-01-13:
Update: used nginx proxypass - I now have a functioning webserver! :) I'm going to go ahead and close this. Thanks for your help!

ChadOhman, 2020-01-13:
f

ChadOhman, 2020-01-13:
Just kidding - I can't view the table - the hyperlink is pointed to http://server/table - not table.html. pointing to table.html gives me a 404 though the file exists in var/www/. Interesting.

SA7BNT, 2020-01-13:
OK. The config on my side is pointing at /var/www/ 
Take a look inside autowx2 there you will see a map var/www/
Inside this there will be your DashBoard. I just pointing symlinks to this map and all was running fine at my side. 

> sa7bnt@owncloud:/var/www/html/autowx2$ ls -lisa
insgesamt 12
390981 4 drwxrwxr-x  3 www-data www-data 4096 Apr 19  2019 .
390984 4 drwxrwxr-x 44 www-data www-data 4096 Jan 13 02:00 ..
398943 0 lrwxrwxrwx  1 www-data www-data   32 Nov 21  2018 css -> /home/sa7bnt/autowx2/var/www/css
390986 0 lrwxrwxrwx  1 www-data www-data   39 Nov 21  2018 index.html -> /home/sa7bnt/autowx2/var/www/index.html
390991 0 lrwxrwxrwx  1 www-data www-data   38 Nov 21  2018 index.tpl -> /home/sa7bnt/autowx2/var/www/index.tpl
398891 0 lrwxrwxrwx  1 www-data www-data   42 Nov 21  2018 nextpass.html -> /home/sa7bnt/autowx2/var/www/nextpass.html
398893 0 lrwxrwxrwx  1 www-data www-data   41 Nov 21  2018 nextpass.png -> /home/sa7bnt/autowx2/var/www/nextpass.png
398901 0 lrwxrwxrwx  1 www-data www-data   46 Nov 21  2018 noaa_dirlist.html -> /home/sa7bnt/autowx2/var/www/noaa_dirlist.html
398921 0 lrwxrwxrwx  1 www-data www-data   39 Nov 21  2018 recordings -> /home/sa7bnt/autowx2/var/www/recordings
398927 0 lrwxrwxrwx  1 www-data www-data   39 Nov 21  2018 table.html -> /home/sa7bnt/autowx2/var/www/table.html
415504 4 drwxr-xr-x  2 sa7bnt   sa7bnt   4096 Apr 19  2019 wiki_bilder
sa7bnt@owncloud:/var/www/html/autowx2$

Eric738, 2020-08-20:
hello Chad
when you use nginx reverse onwhich file or directory are you pointing to in the nginx configuration file? and flask becomes what? Thank you for the info

eric