filipsPL/autowx2 #93: Porting to python3 and Debian buster planned?

dg9vh, 2020-02-29:
Hi,

as Python2.7 is out of maintainance since 01.01.2020, are there some plans to port it to python3 and make it runable with Debian buster?

filipsPL, 2020-03-02:
@dg9vh thanks for your post.

The fact is that python2 is still alive. I have to check this out, but I guess we would have to migrate not only the code of the program itself, but also all dependencies and libraries used. This moght be a lot of work.

But python2 is still available, so I'm sure you can still install the program on any modern linux system. You can also use conda (miniconda) for setting up the environment.

If you have problems with this, please report it here.

dg9vh, 2020-03-02:
I got some issues by using install.sh-script... on raspbian buster it would not install - but when installing the dependencies via apt, it is working now. Only flask-packets I had to install via pip because of no deb-package...

All ok now, I close this issue.

About python you are right - it is still running but in next debian/raspbian-version python2.7 would not be part of the distribution because end of life. You can find some info about this at: https://dev.to/scriptautomate/python-2-2020-end-of-life-eol-and-what-it-means-30p2

But after all: Issue is closed, all up and running :-) That's the plot :-)

closed 2020-03-02 by dg9vh