filipsPL/autowx2 #95: Having a problem getting autowx2.py to run:

ZL1RAH, 2020-04-23:
Having a problem getting autowx2.py to run:
pi@raspberrypi:~/autowx2 $ ./autowx2.py - It looks like something with how matplotlib might be installed, and not sure what to do here. Can you help? Error codes below:
------------------------------------
Traceback (most recent call last):
  File "./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/pi/autowx2/autowx2_functions.py", line 26, in <module>
    import matplotlib
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/__init__.py", line 133, in <module>
    from matplotlib.rcsetup import defaultParams, validate_backend, cycler
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/rcsetup.py", line 31, in <module>
    from matplotlib.fontconfig_pattern import parse_fontconfig_pattern
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/fontconfig_pattern.py", line 28, in <module>
    from backports.functools_lru_cache import lru_cache
ImportError: No module named functools_lru_cache

filipsPL, 2020-04-23:
Hi, thanks for this report. However it seems there is a problem with matplotlib, please see this thread: https://github.com/matplotlib/matplotlib/issues/9344 - maybe you could follow advises mentioned there (like re-installing matplotlib, or downgrading it)? 

Anyway, please share your progress in solving this problem here :)

ZL1RAH, 2020-04-23:
> matplotlib, please see this thread: matplotlib/matplotlib#9344

Hello
Thank you for the reply, but I managed to get that going by typing "sudo 
apt-get install python-matplotlib"
However I'm stuck because the rest seems to be running, expect the flask 
server. I don't see any web pages, and when I change the port number, 
the startup crashes:
This is my startup:

2020-04-23 09:57  ⚡ Program start
2020-04-23 09:57  Killing all remaining rtl_* processes...
WebSocket transport not available. Install eventlet or gevent and 
gevent-websocket for improved performance.
2020-04-23 09:57  Reading the default dongle shift
  * Serving Flask app "autowx2_functions" (lazy loading)
  * Environment: production
    WARNING: This is a development server. Do not use it in a production 
deployment.
    Use a production WSGI server instead.
  * Debug mode: off
2020-04-23 09:57  Using the default dongle shift: 0 ppm
  * Running on http://127.0.0.1:5010/ (Press CTRL+C to quit) *<----- so 
called running*
2020-04-23 09:57   2. discard NOAA-19, keep ISS
2020-04-23 09:57  Recalculating the new pass table and saving to disk.

How do I get the flask to run?

SA7BNT, 2020-04-23:
> WebSocket transport not available

Hallo from Sweden to New Zealand,

please try to see how it will change the startup if you install
pip install gevent
It seems that this is missing.

73
Tim de SA7BNT

ZL1RAH, 2020-04-23:
Hello Tim 

Thanks for the email. 

I had already done that command so I've tried it again and no fix. I
should be able to see from another computer?? 

I get this exception message: Should I upgrade pip?? 

-------------------------------------------------------- 

Exception:
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/pip/basecommand.py",
line 215, in main
    status = self.run(options, args)
  File
"/home/pi/.local/lib/python2.7/site-packages/pip/commands/install.py",
line 342, in run
    prefix=options.prefix_path,
  File "/home/pi/.local/lib/python2.7/site-packages/pip/req/req_set.py",
line 784, in install
    **kwargs
  File
"/home/pi/.local/lib/python2.7/site-packages/pip/req/req_install.py",
line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File
"/home/pi/.local/lib/python2.7/site-packages/pip/req/req_install.py",
line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/home/pi/.local/lib/python2.7/site-packages/pip/wheel.py", line
345, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/home/pi/.local/lib/python2.7/site-packages/pip/wheel.py", line
323, in clobber
    shutil.copyfile(srcfile, destfile)
  File "/usr/lib/python2.7/shutil.py", line 97, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied:
'/usr/local/lib/python2.7/dist-packages/greenlet.so'
You are using pip version 9.0.1, however version 20.0.2 is available.
You should consider upgrading via the 'pip install --upgrade pip'
command. 

-------------------------------------------------------- 

I ALSO GET THIS RANDOM MESSAGE: 

2020-04-23 21:00  Using the good old dongle shift (a): 0 ppm
2020-04-23 21:00  We have still 52:24s free time to the next pass. Let's
do something useful!
2020-04-23 21:00  Running: /home/pi/autowx2/bin/rtl_433.sh for 52:24s
['/home/pi/autowx2/bin/rtl_433.sh', '3144', '0']
/home/pi/autowx2/
ZL1RAH
fatal: No names found, cannot describe anything.
timeout: failed to run command 'rtl_433': No such file or directory 

I don't know where to fix this? 

Is it alright that I used Raspbian Lite?? 

Cheers for now and thanks again:

Kind Regards,
Rodger Hanson
Tel: +64 9 278-8683
Mob: +64 27 5687659
 Email: rodger_hanson@slingshot.co.nz
 Web: https://zl1rah.blogspot.co.nz/ [3]

SA7BNT, 2020-04-23:
> Permission denied

**Permission denied**
Try to run pip as sudo if you install something.
And yes why not upgrade but run it as i wrote. Use Sudo or command "su -" and then as root 

Best

ZL1RAH, 2020-04-24:
Hello there
OK I had to do the following to get gevent to install:
It was missing the ffi library

#sudo apt-get install libffi-dev
#sudo pip install gevent

However I still get this error:
[2020-04-24 01:52:56,026] WARNING in __init__: WebSocket transport not 
available. Install gevent-websocket for improved performance.

SA7BNT, 2020-04-24:
Hey Rodger, 

that is not an error just a warning.
You may try to give this a try.
[https://pypi.org/project/gevent-websocket/](url)
Use sudo or login as root.

Best

ZL1RAH, 2020-04-24:
Hello
Thanks for the help so far. I think its mostly working, except that I 
cannot see the flask web server from any other computer on my network. I 
can see it from the Rasp PI. Is there a way to change this and how?
It's only looking at 127.0.0.1.  I want to look at 192.168.1.3 on my 
network.

Kind Regards,
Rodger Hanson [ZL1RAH]
Tel: +64 9 278-8683
Mob: +64 27 5687659
Email: rodger_hanson@slingshot.co.nz
Web: https://zl1rah.blogspot.co.nz/
QRZ: https://qrz.com/db/zl1rah

/Please consider the environment before printing this email/

SA7BNT, 2020-04-25:
Hallo Rodger,

im not using flask just because of that. I was getting the same problem and could not figure out how to get access to flask from the locale network or give access to public. 
Im using Apache2. Makes some symlinks to my locale webserver and done.

So im sorry at this point i cant help you. 

Best from Sweden 

Take care of you

ZL1RAH, 2020-04-25:
Hello All
I did some reseach and found this:

Change the flask IP from 127.0.0.1 to 0.0.0.0 so that all machines on 
that net can see the web server

*_Change: _* /usr/local/lib/python2.7/dist-packages/flask_socketio/ 
*__init__.py *to host='0.0.0.0'.

ZL1RAH, 2020-04-25:
Hello All
I wonder if somebody ca tell me if this is usual or not: while running 
the autowx2.py this comes up.

2020-04-25 13:37  Running: /home/pi/autowx2/bin/rtl_433.sh for 144:09s
['/home/pi/autowx2/bin/rtl_433.sh', '8649', '0']
/home/pi/autowx2/
ZL1RAH
fatal: not a git repository (or any of the parent directories): .git    
         <-- whats wrong here
timeout: failed to run command ‘rtl_433’: No such file or directory    
<-- whats wrong here
2020-04-25 14:12   2. discard NOAA-19, keep ISS

SA7BNT, 2020-04-25:
You may missing rtl_433
https://github.com/merbanan/rtl_433

filipsPL, 2020-04-27:
@ZL1RAH rtl_433 is not required to run autowx2. Could you share your config file `/autowx2_conf.py`? Maybe you have it enabled by default in:

`scriptToRunInFreeTime = baseDir + "bin/rtl_433.sh"`

just comment that line and uncomment the follwing:

`scriptToRunInFreeTime = False`

best,
Filip

Eric738, 2020-06-06:
gm 
i have too problemes when i send cmd ./autowx2.py 👍 
2020-06-06 09:45  ⚡ Program start  
2020-06-06 09:45  Killing all remaining rtl_* processes...  
WebSocket transport not available. Install eventlet or gevent and gevent-websocket for improved performance.
2020-06-06 09:45  Reading the default dongle shift  
 * Serving Flask app "autowx2_functions" (lazy loading)
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
2020-06-06 09:45  Recently used dongle shift is: 0.0 ppm  
 * Running on http://127.0.0.1:5010/ (Press CTRL+C to quit)
2020-06-06 09:45   2. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-19, keep ISS  
2020-06-06 09:45   1. discard NOAA-19, keep ISS  
2020-06-06 09:45   2. discard NOAA-15, keep ISS  
2020-06-06 09:45   1. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-19, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-18, keep METEOR-M2  
2020-06-06 09:45  Recalculating the new pass table and saving to disk.  
2020-06-06 09:45   2. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-19, keep ISS  
2020-06-06 09:45   1. discard NOAA-19, keep ISS  
2020-06-06 09:45   2. discard NOAA-15, keep ISS  
2020-06-06 09:45   1. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-15, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-19, keep METEOR-M2  
2020-06-06 09:45   1. discard NOAA-18, keep METEOR-M2  
2020-06-06 09:45   2. discard NOAA-18, keep METEOR-M2  
2020-06-06 09:45   2. discard NOAA-18, keep METEOR-M2  
2020-06-06 09:45   2. discard NOAA-19, keep ISS  
2020-06-06 09:45   2. discard NOAA-18, keep METEOR-M2  
2020-06-06 09:45   2. discard NOAA-18, keep Radiosonde  
2020-06-06 09:45   2. discard NOAA-18, keep Radiosonde  
This is a miracle!
2020-06-06 09:45  Next five passes:  
2020-06-06 09:45  ●    NOAA-18 :: 2020-06-06 10:20 to 2020-06-06 10:34, dur: 14:07, max el. 27°; azimuth: 94° (E→) f=137912500Hz; Decoding: modules/noaa/noaa.sh  
2020-06-06 09:45  ●  METEOR-M2 :: 2020-06-06 10:47 to 2020-06-06 11:01, dur: 14:13, max el. 39°; azimuth: 291° (W←) f=137900000Hz; Decoding: modules/meteor-m2/meteor.sh  
2020-06-06 09:45  ●    NOAA-18 :: 2020-06-06 12:01 to 2020-06-06 12:15, dur: 14:37, max el. 42°; azimuth: 292° (W←) f=137912500Hz; Decoding: modules/noaa/noaa.sh  
2020-06-06 09:45  ● Radiosonde :: 2020-06-06 12:20 to 2020-06-06 14:20, dur: 120:00, max el. 0°; azimuth: 0° (N↑) f=98796500Hz; Decoding: modules/radiosonde/run_radiosonde_scanner.sh  
2020-06-06 09:45  ●        ISS :: 2020-06-06 14:20 to 2020-06-06 14:30, dur: 10:13, max el. 83°; azimuth: 145° (SE↘) f=143625000Hz; Decoding: modules/iss/iss_voice_mp3.sh  
2020-06-06 09:45  Next pass:  
2020-06-06 09:45  ●    NOAA-18 :: 2020-06-06 10:20 to 2020-06-06 10:34, dur: 14:07, max el. 27°; azimuth: 94° (E→) f=137912500Hz; Decoding: modules/noaa/noaa.sh  
2020-06-06 09:45  Killing all remaining rtl_* processes...  
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/autowx2/autowx2_functions.py", line 722, in mainLoop
    while not runTest():
  File "/home/pi/autowx2/autowx2_functions.py", line 268, in runTest
    stderr=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
--------------

i follow infos which are in this thread but havnt results, have you any idea ?
Thank
Eric F1SMV

SA7BNT, 2020-06-06:
If it is a fresh install try to run wxtoimg by hand and accept the license. 

73
SA7BNT de Tim

Eric738, 2020-06-06:
thank Tim, i thought was fix but run in loop 👍 in the window :
43625000Hz; Decoding: modules/iss/iss_voice_mp3.sh  
2020-06-06 16:06  Killing all remaining rtl_* processes...  
2020-06-06 16:06  !! Recording ●        ISS :: 2020-06-06 15:57 to 2020-06-06 16:07, dur: 09:41, max el. 24°; azimuth: 344° (NNW↑↖) f=143625000Hz; Decoding: modules/iss/iss_voice_mp3.sh  
La
fileNameCore=20200606-1557_ISS
satellite=ISS
start=1591451872
azimuth=344
/home/pi/autowx2/
duration=93
peak=24
freq=143625000
Recording to mp3
---------
and close by 👍 
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/autowx2/autowx2_functions.py", line 722, in mainLoop
    while not runTest():
  File "/home/pi/autowx2/autowx2_functions.py", line 268, in runTest
    stderr=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
---------------------------------------------------------------------

thank Eric 73

SA7BNT, 2020-06-06:
Hey Eric,
can you please check that all of the
https://github.com/filipsPL/autowx2/blob/master/requirements.txt
is being installed.

Best

Eric738, 2020-06-07:
Bonjour Tim soon the End hi,..
----------------------
 sudo ./autowx2.py
Traceback (most recent call last):
  File "./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/pi/autowx2/autowx2_functions.py", line 36, in <module>
    from flask_socketio import SocketIO, emit
ImportError: No module named flask_socketio
--------------------------------------------
i dont understand how install flack..

Thank in advance
73 Eric

SA7BNT, 2020-06-07:
Yes! https://pypi.org/project/Flask-SocketIO/

Eric738, 2020-06-07:
it would have been too easy...Houston we have a probléme :)
i installed python 3.8.0 
-----------------------------------------------------
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/autowx2/autowx2_functions.py", line 722, in mainLoop
    while not runTest():
  File "/home/pi/autowx2/autowx2_functions.py", line 268, in runTest
    stderr=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
---
i saw in other thead one people installed nginx reverse proxy..after this error but after read doc seems hard to write the setup