filipsPL/autowx2 #96: radiosonde_auto_rx

OK1SLM, 2020-05-23:
2020-05-23 11:00:41,908 INFO:APRS-IS - Uploaded to APRS-IS: OK1SLM-15>APRARX,SONDEGATE,TCPIP,qAR,OK1SLM-15:!5045.51N/01518.24E`radiosonde_auto_rx SondeGate v1.3.1 !w!R!

2020-05-23 11:02:56,236 INFO:Scanner #00000002 - Running frequency scan.
2020-05-23 11:03:16,362 INFO:Scanner #00000002 - Detected peaks on 19 frequencies (MHz): [ 403.2   401.65  405.5   402.46  401.93  401.49  400.87  405.58  405.64
  401.    401.61  405.88  400.69  401.55  404.9   404.23  404.97  403.19
  404.84]
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 596, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 331, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/SocketServer.py", line 654, in __init__
    self.finish()
  File "/usr/lib/python2.7/SocketServer.py", line 713, in finish
    self.wfile.close()
  File "/usr/lib/python2.7/socket.py", line 283, in close
    self.flush()
  File "/usr/lib/python2.7/socket.py", line 307, in flush
    self._sock.sendall(view[write_offset:write_offset+buffer_size])

Hi,
It's a problem or should I ignore it.

filipsPL, 2020-05-25:
Are you intended to run `radiosonde` in a background? If no, remove the radiosonde fragment from your config file.

OK1SLM, 2020-05-25:
Yes, the radiosonde are running and I want them to run

filipsPL, 2020-05-26:
Does radiosonde itself runs without problems?

OK1SLM, 2020-05-26:
2020-05-26 16:51:13,294 INFO:Scanner #00000004 - Detected peaks on 6 frequencies (MHz): [ 403.2   401.65  401.19  401.3   401.93  405.49]
2020-05-26 16:52:28,216 INFO:Scanner #00000004 - Running frequency scan.
2020-05-26 16:53:28,256 CRITICAL:Scanner #00000004 - rtl_power call failed with return code 137.
2020-05-26 16:53:28,258 CRITICAL:Scanner #00000004 - rtl_power reported error: Found 3 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000003
  1:  Realtek, RTL2838UHIDIR, SN: 00000004
  2:  Realtek, RTL2838UHIDIR, SN: 00000002

Using device 1: Generic RTL2832U OEM
Number of frequency hops: 3
Dongle bandwidth: 2479166Hz
Downsampling by: 1x
Cropping by: 20.00%
Total FFT bins: 12288
Logged FFT bins: 9830
FFT bin size: 605.27Hz
Buffer size: 16384 bytes (3.30ms)
Reporting every 20 seconds
Found Rafael Micro R820T tuner
Tuner gain set to automatic.
Exact sample rate is: 2479166.000334 Hz
[R82XX] PLL not locked!
Signal caught, finishing scan pass.
Killed

2020-05-26 16:53:28,260 WARNING:Scanner #00000004 - RTLSDR produced no output... resetting and retrying.
2020-05-26 16:53:28,823 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/006
2020-05-26 16:53:39,068 INFO:Scanner #00000004 - Running frequency scan.
2020-05-26 16:53:59,384 INFO:Scanner #00000004 - Detected peaks on 10 frequencies (MHz): [ 403.2   401.65  401.19  401.12  401.2   401.17  401.23  401.24  401.18

Sometimes it crashes but restarts itself.

filipsPL, 2020-05-29:
> Sometimes it crashes but restarts itself

So, if it runs and everything works well, then don't worry!