filipsPL/autowx2 #98: Looking for working Autostart script for raspberrypi p3

v222008, 2020-06-19:
script running well only run as root cd /home/pi/autowx2    and run manualy file autowx2.py

not works on start script

```
#!/bin/bash
cd /home/pi/autowx2/
python autowx2.py  &
```
or 

`/usr/bin/tmux new-session -d 'python /home/pi/autowx2/autowx2.py'`

or 

```
sudo nano /lib/systemd/system/autowx2.service

[Unit]
Description=autowx2
After=multi-user.target
[Service]
Restart=always
RestartSec=10
WorkingDirectory=/home/pi/autowx2
ExecStart=/usr/bin/python /home/pi/autowx2/autowx2.py
[Install]
WantedBy=multi-user.target

sudo chmod 644 /lib/systemd/system/autowx2.service
sudo systemctl daemon-reload
sudo systemctl enable autowx2.service
sudo systemctl start autowx2.service

```

filipsPL, 2020-06-19:
Thanks for this report. What system do you use (and which version)? (answer like "raspbian installed a week ago" will do :) )

v222008, 2020-06-21:
Ok)
my system is 

```
root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
root@raspberrypi:/home/pi#
```
1st method is successful when it runs manually is

`/usr/bin/tmux new-session -d 'python /home/pi/autowx2/autowx2.py'`

in other cases the main file autowx2.py is started but modules scripts (like in  /autowx2/modules ) are not executed..
they are executed but do not read variables
```
echo $fileNameCore >> $logFile
echo $satellite >> $logFile
echo $start >> $logFile
echo $duration >> $logFile
echo $peak >> $logFile
echo $freq >> $logFile
```

p.s.

and in my case the web works correctly with the settings

#wwwDir=baseDir + 'var/www/'
wwwDir='/var/www/wx/'

#wwwRootPath="/"
wwwRootPath=""

Thank you for your attention)

thecivvie, 2020-10-24:
I am having the same issue with my Pi. It starts ok but the scripts don't work correctly. I don't get any images except the neat map. If I log in and run from the command line, there is no issue

SA7BNT, 2020-10-25:
> I am having the same issue with my Pi. It starts ok but the scripts don't work correctly. I don't get any images except the neat map. If I log in and run from the command line, there is no issue

Can you please upload or see something strange in the log??

thecivvie, 2020-10-30:
Seeing nothing in the logs. Unless I log into the Pi and leave it connected , it runs ok. If I started it automatically at boot time it only generates the NO jpg file.

thecivvie, 2020-10-30:
From a failed grab log:
/home/pi/autowx2/
RenvyleWeather
fileNameCore=20201030-1812_NOAA-19
satellite=NOAA-19
peak=73
freq=137100000
dongleIndex=0
**** MCIR-precip
**** HVC
**** MSA
**** therm
**** HVCT-precip
**** NO
**** MCIR-precip
**** MSA
**** therm
Fri 30 Oct 18:27:18 GMT 2020
/home/pi/autowx2/

the previous log was:
3:31, max el. 25°; azimuth: 54° (NE↗) f=137100000Hz; Decoding: modules/noaa/noaa.sh  
RenvyleWeather
fileNameCore=20201030-1632_NOAA-19
satellite=NOAA-19
start=1604075556
peak=25
azimuth=54
freq=137100000
wavrate=11025
dongleIndex=0
**** MCIR-precip
**** HVC
**** MSA
**** therm
**** HVCT-precip
**** NO
**** MCIR-precip
**** HVC
**** MSA
**** therm
**** HVCT-precip
**** NO
Fri 30 Oct 16:49:48 GMT 2020
/home/pi/autowx2/
RenvyleWeather
Fri, 30 Oct 2020 16:49:49 +0000
0

thecivvie, 2020-11-24:
Still not finding it easy to start automatically

SA7BNT Your autowx2 is not updating

SA7BNT, 2020-11-24:
> Still not finding it easy to start automatically
> 
> SA7BNT Your autowx2 is not updating

Hey @thecivvie not updating?? How do you mean? If i looking at my autowx2 all is running fine

thecivvie, 2020-11-24:
It is showing the last update as the 21st no images since
[image]

SA7BNT, 2020-11-24:
@thecivvie damm!! Restarted. Thanks for info