filipsPL/autowx2 #99: New installation and error message

Eric738, 2020-06-20:
GM
after a new installation i meet this probléme 👍 
  
./autowx2.py
Traceback (most recent call last):
  File "./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/pi/autowx2/autowx2_functions.py", line 26, in <module>
    import matplotlib
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/__init__.py", line 133, in <module>
    from matplotlib.rcsetup import defaultParams, validate_backend, cycler
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/rcsetup.py", line 31, in <module>
    from matplotlib.fontconfig_pattern import parse_fontconfig_pattern
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/fontconfig_pattern.py", line 28, in <module>
    from backports.functools_lru_cache import lru_cache
ImportError: No module named functools_lru_cache
-------------------------------------------------------------

how i can fix it ? thank for help

Eric

filipsPL, 2020-06-22:
Thanks for reporting. This is a possible problem with matplotlib. Please look at the solutions mentioned here:

https://stackoverflow.com/questions/47179433/python-2-7-functools-lru-cache-does-not-import-although-installed

Eric738, 2020-06-22:
thank for url, i will follow steps to install. Do you think update the project under python 3.8 one day ? because few messages said while installation that python 2.7 is out of date ? thank infos

filipsPL, 2020-06-22:
> Do you think update the project under python 3.8 one day

yes, this is one of the main goals for the future. Unfortunatelly the problem are some libraries, which are python 2.7 only... But we will get to this point one day :smile:

Eric738, 2020-06-22:
cool, I hope one day because the program is really great I can't wait to get it to work properly, this is not currently the case. 2 concerns 
1 / aprs.sh generates an error message and does not listen to the right frequency on 144800
2 / the website is not working properly, on the first page I have the log as if I were in terminal function instead of displaying the photos received and I cannot solve this problem. where can i look? :(

filipsPL, 2020-07-21:
> 2 / the website is not working properly, on the first page I have the log as if I were in terminal function instead of displaying the photos received and I cannot solve this problem. where can i look? :(

which website are you referring to?

Eric738, 2020-07-28:
site on my raspi..he recovered photos are not displayed we only see the log displayed

OK1SLM, 2024-01-08:
Please heelp 

ok1slm@raspberrypi:~/autowx2 $ ./autowx2.py
Traceback (most recent call last):
  File "./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/ok1slm/autowx2/autowx2_functions.py", line 14, in <module>
    import predict
ImportError: No module named predict

SA7BNT, 2024-01-08:
> Please heelp
> 
> ok1slm@raspberrypi:~/autowx2 $ ./autowx2.py Traceback (most recent call last): File "./autowx2.py", line 13, in from autowx2_functions import * # all functions and magic hidden here File "/home/ok1slm/autowx2/autowx2_functions.py", line 14, in import predict ImportError: No module named predict

pip install PREDICT

OK1SLM, 2024-01-08:
Could not find a version that satisfies the requirement SimpleITK==1.1.0 (from PREDICT) (from versions: 1.0.1, 1.2.0, 2.1.0, 2.1.1.2, 2.3.0)
No matching distribution found for SimpleITK==1.1.0 (from PREDICT)

SA7BNT, 2024-01-08:
We are talking Bullseye?

OK1SLM, 2024-01-08:
$ hostnamectl
   Static hostname: raspberrypi
         Icon name: computer
        Machine ID: c92a5dc2fbb64690ab00553567909537
           Boot ID: 34f4767d0f3d42d3872b7ed460b47304
  Operating System: Raspbian GNU/Linux 10 (buster)
            Kernel: Linux 5.10.103-v7+
      Architecture: arm

SA7BNT, 2024-01-08:
> $ hostnamectl Static hostname: raspberrypi Icon name: computer Machine ID: c92a5dc2fbb64690ab00553567909537 Boot ID: 34f4767d0f3d42d3872b7ed460b47304 Operating System: Raspbian GNU/Linux 10 (buster) Kernel: Linux 5.10.103-v7+ Architecture: arm

Maybe @filipsPL has an answer

SA7BNT, 2024-01-08:
> hostnamectl

sa7bnt@exemple:~$ hostnamectl
   Static hostname: exemple.com
         Icon name: computer-desktop
           Chassis: desktop
        Machine ID: e4f013b5dee44d7091babffb1d48b6f9
           Boot ID: 073517f09400495bbbdbc210b7c98465
  Operating System: Debian GNU/Linux 10 (buster)
            Kernel: Linux 4.19.0-24-amd64
      Architecture: x86-64

[image]

filipsPL, 2024-01-08:
> ok1slm@raspberrypi:~/autowx2 $ ./autowx2.py Traceback (most recent call last): File "./autowx2.py", line 13, in from autowx2_functions import * # all functions and magic hidden here File "/home/ok1slm/autowx2/autowx2_functions.py", line 14, in import predict ImportError: No module named predict

Please install pypredict: `pip install pypredict` (this is different from PREDICT mentioned above) -> https://pypi.org/project/pypredict/

OK1SLM, 2024-01-08:
ok1slm@raspberrypi:~/autowx2 $ ./autowx2.py
Traceback (most recent call last):
  File "./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/ok1slm/autowx2/autowx2_functions.py", line 36, in <module>
    from flask_socketio import SocketIO, emit
  File "/home/ok1slm/.local/lib/python2.7/site-packages/flask_socketio/__init__.py", line 21, in <module>
    import socketio
  File "/home/ok1slm/.local/lib/python2.7/site-packages/socketio/__init__.py", line 3, in <module>
    from .client import Client
  File "/home/ok1slm/.local/lib/python2.7/site-packages/socketio/client.py", line 7, in <module>
    import engineio
  File "/home/ok1slm/.local/lib/python2.7/site-packages/engineio/__init__.py", line 3, in <module>
    from .client import Client
  File "/home/ok1slm/.local/lib/python2.7/site-packages/engineio/client.py", line 2, in <module>
    from json import JSONDecodeError
ImportError: cannot import name JSONDecodeError

OK1SLM, 2024-01-08:
ok1slm@raspberrypi:~/autowx2 $ ./autowx2.py
Traceback (most recent call last):
  File "./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/ok1slm/autowx2/autowx2_functions.py", line 36, in <module>
    from flask_socketio import SocketIO, emit
ImportError: No module named flask_socketio

filipsPL, 2024-01-09:
Please check the [requirements](https://github.com/filipsPL/autowx2/blob/master/requirements.txt):
```
pypredict
matplotlib
numpy
Flask
"Flask-SocketIO<=5.2.0"
```
and pip install any packages you have not yet installed. In this case try:

`pip install Flask-SocketIO==5.2.0`

OK1SLM, 2024-01-09:
ERROR: Could not find a version that satisfies the requirement bidict>=0.21.0 (from python-socketio>=5.0.2->Flask-SocketIO<=5.2.0) (from versions: 0.1.5, 0.2.1, 0.3.0, 0.3.1, 0.9.0rc0, 0.9.0.post1, 0.10.0, 0.10.0.post1, 0.11.0, 0.12.0.post1, 0.13.0, 0.13.1, 0.14.0, 0.14.1, 0.14.2, 0.15.0.dev0, 0.15.0.dev1, 0.15.0rc1, 0.15.0, 0.16.0, 0.17.0, 0.17.1, 0.17.2, 0.17.3, 0.17.4, 0.17.5, 0.18.0, 0.18.1, 0.18.2, 0.18.3, 0.18.4)
ERROR: No matching distribution 

I had to use the version to install     pip install "Flask-SocketIO<5.0.0"
but then there is another problem

ok1slm@raspberrypi:~/autowx2 $ ./autowx2.py
Traceback (most recent call last):
  File "./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/ok1slm/autowx2/autowx2_functions.py", line 36, in <module>
    from flask_socketio import SocketIO, emit
  File "/home/ok1slm/.local/lib/python2.7/site-packages/flask_socketio/__init__.py", line 21, in <module>
    import socketio
  File "/home/ok1slm/.local/lib/python2.7/site-packages/socketio/__init__.py", line 3, in <module>
    from .client import Client
  File "/home/ok1slm/.local/lib/python2.7/site-packages/socketio/client.py", line 7, in <module>
    import engineio
  File "/home/ok1slm/.local/lib/python2.7/site-packages/engineio/__init__.py", line 3, in <module>
    from .client import Client
  File "/home/ok1slm/.local/lib/python2.7/site-packages/engineio/client.py", line 2, in <module>
    from json import JSONDecodeError
ImportError: cannot import name JSONDecodeError

OK1SLM, 2024-01-10:
ok1slm@raspberrypi:~/autowx2 $ ./autowx2.py
Traceback (most recent call last):
  File "/home/ok1slm/autowx2/./autowx2.py", line 13, in <module>
    from autowx2_functions import * # all functions and magic hidden here
  File "/home/ok1slm/autowx2/autowx2_functions.py", line 36, in <module>
    from flask_socketio import SocketIO, emit
  File "/home/ok1slm/.local/lib/python3.9/site-packages/flask_socketio/__init__.py", line 19, in <module>
    from flask import _request_ctx_stack, has_request_context, json as flask_json
ImportError: cannot import name '_request_ctx_stack' from 'flask' (/usr/local/lib/python3.9/dist-packages/flask/__init__.py)