filipsPL/autowx2 #102: METEOR M2 images

ThomasBartleet, 2020-08-09:
Hi all,

I'm getting NOAA images absolutely fine, but I'm getting METEOR images like so:

[image] [image]

Does anyone have any suggestions as to why this is happening and how I can improve this? [This reddit post](https://www.reddit.com/r/amateurradio/comments/67gpsm/meteor_m2_receiving_images_with_black_stripes/) suggest removing `lna4all`, but I can't find any usage of it.

I'm using this [TA-1 cross-dipole antenna](https://www.wimo.com/en/ta-1) to receive my signals, which is connected to my RTL-SDR, which is connected to my RaspberryPi.

TIA!

SA7BNT, 2020-08-09:
Please post your config for M2

Best

ThomasBartleet, 2020-08-09:
@SA7BNT Wow, what a quick response!

There are are few config file for M2: `M2-1-72k.cgf`, `M2-2-72k.cfg`, and `M2-2-80k.cfg`. Below is the contents of `M2-2-80k.cfg`:
```
############## mlrpt RUNTIME CONFIGURATION FILE: mlrptrc ########
#                                                                 #
# PLEASE NOTE! In order to keep the parser for this file simple,  #
# the runtime configuration entries below must all be present and #
# in the order presented, since they are all read-in in a fixed   #
# sequence. Blank lines or those starting with a # (comments etc) #
# are ignored.                                                    #
#                                                                 #
# Please check ALL ENTRIES and EDIT as needed!!                   #
#                                                                 #
###################################################################
#
####### SDR RECEIVER CONFIGURATION #######
# SDR Receiver type to be used for satellite reception.
# Available support is for RTL-SDR and  AIRSPY R2 and Mini
# Default is to use an RTL-SDR device with librtlsdr.
RTL-SDR
#AIRSPY
#
# Device Index. Default is 0 (use first SDR device).
# Maximum value is 8 for data sanity checking.
0
#
# SDR Receiver's I/Q Sample Rate S/s.
# THIS DEPENDS ON THE SDR RX IN USE.
# For RTL-SDR,      288000 S/s.
# For Airspy R2,    2500000 S/s.
# For Airspy Mini,  3000000 S/s.
288000
#2500000
#3000000
#
# Low Pass Roofing Filter Bandwidth. This should be more than
# 100000 Hz for sanity checking. Default is 110000 Hz
# and this is probably the narrowest useable bandwidth.
110000
#
# Manual Tuner Gain setting. If 0, then Auto AGC will be used.
# Value range is 0 - 100% and default is 95% (manual AGC used).
95
#
# Synthesizer Frequency Error correction factor. This is in
# ppm (parts per million) but it has to be an integer number.
# Maximum +/- value for sanity checks is 100.
0
#
####### Mlrpt operational parameters #######
#
# Satellite Transmitter Frequency in kHz (default Meteor M2)
137900
#
# Duration in seconds of image decoding operation (900). This
# is usually to long (15 min) unless a high gain rotating
# directional antenna is used. This value can be changed here
# or using the "Operation Timer" menu item in the pop-up menu.
900
#
####### LRPT Demodulator Parameters #######
#
# RRC (Raised Root Cosine) Filter Order. Default is 32.
32
#
# RRC Filter alpha factor. Default is 0.6 and need not be changed.
0.6
#
# Initial Costas PLL Loop Bandwidth in Hz. Default is 200. May be
[... 90 lines cut ...]
# Blue Channel image that are watery areas (seas or lakes).
#
# Blue Channel minimum pixel value. Must be between 0-255.
# It is also the Blue Channel pixel value below which pixel
# values are mapped into the range between these two values.
60
#
# Blue Channel maximum pixel value when mapping pixel values
# into the range between these two values. Must be higher
# than the above minimum but less than 255.
80
#
# Blue channel cloudy area pixel value threshold. In pseudo-
# colorize mode, pixels with a value higher than this will be
# assumed to be cloudy areas and the RGB values in the combo
# color image will be made all equal to the blue channel pixels
210
#
# Directory to save received images in. Normally it would be
# /home/<user>/mlrpt/images/ if Default is entered here. Else
# the specified directory will be used. Maximum 80 characters.
Default (/home/<user>/mlrpt/images/)
#/var/lrpt/images/
#
```

SA7BNT, 2020-08-09:
> @SA7BNT Wow, what a quick response!
>
> There are are few config file for M2: `M2-1-72k.cgf`, `M2-2-72k.cfg`, and `M2-2-80k.cfg`. Below is the contents of `M2-2-80k.cfg`:
>
> ```
> ############## mlrpt [...]

OK, you may just use M2-1-72k.cgf. M2-2* is already dead :( 
Here is my conf

[Meteor_M2_conf.txt](https://github.com/filipsPL/autowx2/files/5047275/Meteor_M2_conf.txt)

Here can you get the some more information
http://happysat.nl/Meteor/html/Meteor_Status.html?fbclid=IwAR2HdqUTTYDDi3unstM7hziPGa105udOXDn9GIim6wFYNlfIFPDv5uvaIW8

ThomasBartleet, 2020-08-09:
Ah shoot! Ok, I will try with your config. Many thanks! The next fly-over is at 8pm (BST), so I will wait to see how that works out.

SA7BNT, 2020-08-09:
You welcome!! ;)

SA7BNT, 2020-08-09:
BTW @ThomasBartleet my minElev = 30
For me is all under 30° is crap

ThomasBartleet, 2020-08-10:
Hi @SA7BNT , I tried with your config file, but I still got some the following picture:
[image]

I have my min elavation at 50 degrees.

What is the main cause for so much data loss? Do I need an amplifier between my antenna and my SDR? Increase the gain? Or is it just my antenna that is just pants?

My next fly-over is now not until tomorrow morning. I'll record the audio and run some tests on the recorded audio. Maybe then I'll be able to tweak the configuration to best suit for me. Or maybe I'll find there's an issue somewhere else.

ThomasBartleet, 2020-08-10:
It looks like there might be some insight over on [Reddit](https://www.reddit.com/r/RTLSDR/comments/8xlhrq/can_anyone_help_decoding_my_meteor_m2_recording/). I'll have a look at this at some point, but it seems to be hardware-related issues on the receiving end. In other words, my equipment is lacking. Thoughts?

SA7BNT, 2020-08-10:
@ThomasBartleet i´m just running an SDR-RTL.com 50m 50 Ohm Coax (Aircell7) and the antenna is a homemade turnstile on hardware side. Is the RTL-SDR a known like SDR-COM or Nooelec?? Is the RTL-SDR running in idle time?? Asking because if not the RTL-SDR  getting cold and then the ppm will drift away. Hope you understand. 
BTW you can also catch me on Telegram if you are interested. @SA7BNT 
What language are you?

Best

ThomasBartleet, 2020-08-20:
Hi @SA7BNT !
Sorry for getting back to you late! I've been really busy with other commitments and finding alternatives to MLRPT. It turns out there are indeed other alternatives, especially for the Raspberry Pi. Below is an image from yesterday evening:
[image]

I've got another pass later this morning, which should get me a better day-time picture.

From this, it looks like that the MLRPT that AutoWX2 uses is too heavy for the RaspberryPi. I ended up using [`meteor_demod`](https://github.com/dbdexter-dev/meteor_demod) and [`meteor_decoder`](https://github.com/artlav/meteor_decoder), and I re-wrote the `meteor_record.sh` script that allows the RPi to get the image above. Maybe this needs a PR? I would have thought if it can run on the RPi, it can run on bigger (and better) machines?

Just to answer your questions: I am using a NooElec dongle, and the program does not run anything while waiting for passes (maybe the PPM does drift a bit?).

ThomasBartleet, 2020-08-20:
Just to show the effectiveness of the changes I made, below is the image from this morning:
[image]

This beats the original images I was getting (example in the initial comments way above) using the current MLRPT. @filipsPL , is this worth looking into, or is it too much of an isolated case?

0nnyx, 2020-12-27:
@ThomasBartleet Could you share your meteor script & config ? Also running on Rpi and would like to give meteor another try.

ThomasBartleet, 2020-12-29:
@0nnyx Sure! You can find the actual processing code on my fork over at https://github.com/ThomasBartleet/autowx2/pull/1. You'll find I've retained a similar structure as the NOAA processing files.

For additional features, I've had to modify the actual `meteor.conf` file a little bit. It should be self-explanatory though:
```
#
# meteor m2 module configuration file
#

medetExec="$baseDir/medet/medet_arm"

# directory with meteor stuff
meteorDir="$recordingDir/meteor"

# directory of meteor recordings
rootMeteorRecDir="$meteorDir/rec"
recdir="$rootMeteorRecDir/"$(date +"%Y/%m/%d")

# directory where the images finally will go
rootMeteorImgDir="$meteorDir/img"
imgdir="$rootMeteorImgDir/"$(date +"%Y/%m/%d")

# Option to save images as PNG or JPG (if neither, defaults to JPG)
imageExtension="png"
# imageExtension="jpg"

# resize images to the given size to avoid growing of the repository; in px
# resizeimageto=1024

# Shall we remove the audio and processing files? If not, comment out below
# or leave blank.
removeFiles="true"

# The images stored take up room, as well as the recordings (if storing recordings). Over time,
# this data might not be necessary, so it should to be removed. Store the data for a specificed
# number of days (set to 0 for "do not remove").
keepDataForDays=7
```

Just a note, I'm not storing the images for each individual channel. The script only gets the image from combining the channels.

0nnyx, 2020-12-29:
Thanks, will see how it performs :) Your Meteor branch also has that nice script for removing files older than X days, which is a must on Rpi with limited SD space.

0nnyx, 2021-03-15:
@ThomasBartleet The default RGB values for medet on your repo systematically generate very red images. I guess you found a way to improve that. Just in case, I altered a bit the meteor_process.sh to have false colored during daylight (64 & 65 only) and infrared (68 only) when sun is down:
```
# Decide the file and make image.
if [ -f $decodedAudioFile ]; then
    echo "I got a successful ${3}.dec file. Creating false color image"
    if [ $(date +%H) -lt 17 ]; then
      ./meteor_decoder/medet $decodedAudioFile "${imageFile}-122" -r 65 -g 65 -b 64 -d
    else
      ./meteor_decoder/medet $decodedAudioFile "${imageFile}-122" -r 68 -g 68 -b 68 -d
    fi
    convert "${imageFile}-122.bmp" "${imageFile}.jpg"
    if [ -f "${imageFile}.jpg" ]; then
      rm "${imageFile}-122.bmp"
    fi
```

ThomasBartleet, 2021-03-15:
@0nnyx Yes, I realised they were a bit red-ish, and I was going to look into it, but never got around it. Your bit of code looks promising! It'll definitely help get night images as well as day images. I've got a pass tomorrow morning, so I will see how that works out.

Ultimately, maybe we can use some open-source sunrise/sunset API (for example, [this one](https://sunrise-sunset.org/api)) to determine when a user will be in the dark when the M2 satellite is overhead (`#TODO`).

filipsPL, 2021-03-16:
> Ultimately, maybe we can use some open-source sunrise/sunset API (for example, [this one](https://sunrise-sunset.org/api)) to determine when a user will be in the dark when the M2 satellite is overhead (`#TODO`).

We may use a self hosted solution, such as astral:
https://pypi.org/project/astral/1.2/
https://astral.readthedocs.io/en/latest/#sun

ploeffler, 2021-03-16:
maybe an offline-solution for that would be better: predict can do that

regards peter
--
Disclaimer: May be opened and read by the NSA, CIA, GCHQ, KGB, BND, and
whoever else it may NOT concern.

0nnyx, 2021-04-01:
Meteor will only show visible channels until September - see http://happysat.nl/Meteor/html/Meteor_Status.html - so best to keep medet with the default RGB values 66 65 64 and expect all evening images to be useless for the next 5 months.
Would be good to have autowx2_conf with a priority based on the time of the day

ThomasBartleet, 2021-04-02:
@0nnyx Good shout! That should make things much easier (and better coloured) for now.

0nnyx, 2023-07-29:
In case anyone still has Meteor in their wanted passes, only the new N2-3 satellite is active - see happysat.nl page for details.
Note the antenna of this new satellite didn't manage to deploy fully. Therefore recordings won't be as successful as old Meteor.

In terms of decoding, mlrpt software is outdated, extremely slow (taking ~1hour on Pi3) and doesn't produce the best results.
Best to use Satdump from https://github.com/SatDump/SatDump or meteor_demod + meteor_decode from https://github.com/dbdexter-dev. Both take ~5min on Pi3.

Satdump produces the best results but isn't suited for automated gallery processing. 
CLI examples (GUI also available) : 
```
satdump meteor_m2-x_lrpt baseband <wav file path> <output directory> (for 72k)
satdump meteor_m2-x_lrpt_80k baseband <wav file path> <output directory> (for 80k)
```
Meteor_demod & decode is easy to integrate for automated gallery.
- 72k requires these commands : 
```
meteor_demod -B -m oqpsk -o $demodulatedAudioFile $normalisedAudioFile
meteor_decode -B -d -q -o "${imageFile}.bmp" $demodulatedAudioFile
```
- 80k requires those commands : 
```
meteor_demod -B -m oqpsk -r 80000 -o $demodulatedAudioFile $normalisedAudioFile
meteor_decode -B -d -i -q -o "${imageFile}.bmp" $demodulatedAudioFile
```

OK1SLM, 2024-01-12:
Does anyone have the best current settings on Meteor?

0nnyx, 2024-01-12:
> Does anyone have the best current settings on Meteor?

Don't really understand your question... Read my post above first.
Current (and past months) settings for N2-3 are 72K for symbol rate and frequency of 137900000Hz.
Same situation as before regarding image retrieval depending on software : satdump > meteor_demod + meteor_decode > mlrpt
If you have an external antenna on your roof, you can probably get decent images despite the bad antenna deployment.
Having my antenna stuck to a window in the attic, I rarely get any good image unlike the previous dead meteor - attaching samples.
Dead M2-1 : 
[image]

Current M2-3 : 
[image]

OK1SLM, 2024-01-12:
Sorry, I didn't notice the new post.
Can you give me the cfg mrlpt ?

OK1SLM, 2024-01-30:
HI Onnyx,
I installed your modified version and I want to ask if receiving images from Meteor M2-3 is functional?
https://www.ok1slm.cz/

OK1SLM, 2024-01-31:
I have a QFH antenna with a preamplifier, but the program keeps restarting when I receive Meteor, I have no idea if I should edit something in the script.

0nnyx, 2024-02-01:
> I have a QFH antenna with a preamplifier, but the program keeps restarting when I receive Meteor, I have no idea if I should edit something in the script.

Here are my scripts, requiring meteor_demod + meteor_decode from https://github.com/dbdexter-dev
I won't help you further though.
Use the Satdump windows installer with the GUI if the CLI automated linux is too complex.

**modules/meteor-m2/meteor_record.sh**
```
#!/bin/bash
startT=$(date +%H%M -d "$DATE + 1 min" -u)
stopT=$(date +%H%M -d "$DATE + $duration sec" -u)
durationMin=$(bc <<< "$duration/60 +2")

#
# recording
#
echo "$startT-$stopT, duration: $durationMin min"
# $1 = Satellite Name
# $2 = Frequency
# $3 = FileName base
# $4 = TLE File
# $5 = EPOC start time
# $6 = Time to capture
# $7 = Satellite max elevation

echo "Satellite Name=$1"
echo "Frequency=$2"
echo "FileName base=$3"
echo "TLE File=$4"
echo "EPOC start time=$5"
echo "Time to capture=$6"
echo "Satellite max elevation=$7"

rawDirectory="${recdir}/raw"
rawAudioFile="${rawDirectory}/${3}.wav"
processedFile="${recdir}/${3}"
normalisedAudioFile="${processedFile}.wav"
demodulatedAudioFile="${processedFile}.qpsk"
decodedAudioFile="${processedFile}.dec"

# Create folders if they don't exist
mkdir -p $rawDirectory

echo "Starting rtl_fm record"
timeout ${6} rtl_fm -f ${2} -M raw -s 288k -g 49.6 -d 1 -A fast -F 9 -E dc -E deemp | sox -t raw -r 288k -c 2 -b 16 -e s - -t wav $rawAudioFile rate 96k

echo "Normalization in progress"
sox $rawAudioFile $normalisedAudioFile gain -n

# Remove the raw audio file. We don't need it anymore, as we can just
# use the normalised audio file.
rm "${recdir}/raw/${3}.wav"
```

**modules/meteor-m2/meteor_process.sh**
```
#!/bin/bash

processedFile="${recdir}/${1}"
imageFile="${imgdir}/${1}"
normalisedAudioFile="${processedFile}.wav"
demodulatedAudioFile="${processedFile}.qpsk"
decodedAudioFile="${processedFile}.dec"

# Demodulate the audio file
echo "Demodulation in progress (QPSK)"
meteor_demod -B -m oqpsk -o $demodulatedAudioFile $normalisedAudioFile
#meteor_demod -B -m oqpsk -r 80000 -o $demodulatedAudioFile $normalisedAudioFile

# Decode th demoodulated file
echo "Decoding in progress (QPSK to BMP)"

meteor_decode -B -d -q -o "${imageFile}.bmp" $demodulatedAudioFile
#meteor_decode -B -d -i -q -o "${imageFile}.bmp" $demodulatedAudioFile
convert -quality 95 "${imageFile}.bmp" "${imageFile}.jpg"
    if [ -f "${imageFile}.jpg" ]; then
      rm "${imageFile}.bmp"
    fi
else
    echo "[DEBUG] Meteor Decoding failed, either a bad pass/low SNR or a software problem"
fi

#if [ "${removeFiles}" -ne "" ]; then
rm $normalisedAudioFile
rm $demodulatedAudioFile
#fi
```

0nnyx, 2024-03-09:
For anyone interested, a new meteor satellite METEOR-M2-4 has been launched and is currently in testing phase.
It's very possible that M2-4 had a proper antenna deployment and would produce good images unlike M2-3.
Since happysat website is down, you can get more info at https://usradioguy.com/meteor-satellite/

SA7BNT, 2024-03-09:
> Here are my scripts, requiring meteor_demod + meteor_decode from https://github.com/dbdexter-dev I won't help you further though. Use the Satdump windows installer with the GUI if the CLI automated [...]

Hey 0nnyx,

did you get img with this setup and autowx2 ?
I´m running your record&process.sh but it is still not possible to get some img

Best

OK1SLM, 2024-03-22:
I'm sad that this project is dead. I switched to the https://github.com/jekhokie/raspberry-noaa-v2 project

filipsPL, 2024-03-22:
The project is not dead. The issue with recording and interpreting METEOR (or any other signal) is out of the scope of this project and the developer.

closed 2024-03-22 by filipsPL