filipsPL/autowx2 #108: Remove Flask

thecivvie, 2021-01-18:
I have tried, and failed, to install on both Debian 10 and 9. The issue comes down to a requirement for bidict for flask_socketio . For python 2, the oldest is 0.18.1 but the standard download requires 0.25.1 I believe.

I tried installing older versions of flask and flask_socketio but that gave a JSONonERROR error. I dont use the webserver that the autowx2 runs with, prefer lighttpd.

What steps do I need to do to remove flask, I am a total noob in  Python

SA7BNT, 2021-01-18:
Cant answer that. But is this a solution??
[https://computingforgeeks.com/how-to-install-pip-2-pip-3-on-debian/](url)

thecivvie, 2021-01-18:
I have 3 different versions now on it. But I can get your git to work ok as there is no flask in it. Just waiting to see what happens with the next pass

SA7BNT, 2021-01-18:
Maybe @filipsPL have a solution for this problem.

filipsPL, 2021-01-18:
I will try to prepare a conda environment which should ease installation of the proper packages.

Another option is to remove flask. It seems that it is not frequently used feature of autowx2...

thecivvie, 2021-01-18:
Thank you. It is definitely the software I want to use. It does everything that is required

EricFROL, 2022-05-31:
Still having this issue over here...

filipsPL, 2025-11-16:
Done.