filipsPL/autowx2 #109: Use noaa-apt instead of wxtoimg

thecivvie, 2021-01-19:
I recently came across the git mentioned in the topic. seems to do everything that wxtoimg does. I am only testing it now so not sure what the images are truly like. Will post one once I do but seems popular on github

thecivvie, 2021-01-19:
[image]
Not a great image as it was a low pass

filipsPL, 2021-01-19:
Thanks for post. We look forward for your long-term observation. You can also send some good images here.

filipsPL, 2023-02-12:
Tested, but so far the results are slightly worse than wxtoimg:

[image]

I will keep track of this software, but now closing the issue. Thanks for reporting!

closed 2023-02-12 by filipsPL