filipsPL/autowx2 #118: Autowx2 file accesing error

davidllorent, 2024-04-27:
2024-04-27 09:07 ● ISS :: 2024-04-27 09:04 to 2024-04-27 09:14, dur: 09:23, max el. 23°; azimuth: 225° (SW↙) f=143625000Hz; Decoding: modules/iss/iss_voice_mp3.sh
2024-04-27 09:07 !! Recording ● ISS :: 2024-04-27 09:04 to 2024-04-27 09:14, dur: 09:23, max el. 23°; azimuth: 225° (SW↙) f=143625000Hz; Decoding: modules/iss/iss_voice_mp3.sh
2024-04-27 09:07 ✖ OS Error during command: modules/iss/iss_voice_mp3.sh 20240427-0904_ISS ISS 1714208693 419 23 225 143625000
2024-04-27 09:07 ✖ OS Error: No such file or directory

I get this error every time. It is running with all permissions.

SA7BNT, 2024-04-27:
What do you get if you just run iss_voice_mp3.sh in its folder?

davidllorent, 2024-04-27:
I've been changing permissions and i get this in its folder:
/home/dvd/autowx2/
fatal: No names found, cannot describe anything.
fileNameCore=20240427-0904_ISS
satellite=ISS
start=1714208693
duration=563
peak=23
azimuth=225
freq=143625000
Recording to mp3
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Tuner gain set to 49.60 dB.
Tuned to 143877000 Hz.
Oversampling input by: 42x.
Oversampling output by: 1x.
Buffer size: 8.13ms
Exact sample rate is: 1008000.009613 Hz
Allocating 15 zero-copy buffers
Sampling at 1008000 S/s.
Output at 24000 Hz.

By the way, the fatal: No names found, cannot describe anything really freaks me out. I dont know if it can cause that problem

SA7BNT, 2024-04-27:
Hmmmm strange.

filipsPL, 2024-04-29:
What happens if you just run `modules/iss/iss_voice_mp3.sh` ?
Do you have `sox` installed? (try running `sox`).

Please provide output of command `ls -la /home/dvd/autowx2/*`